prusa3d/PrusaSlicer #3340: Prusaslicer toobars disappear, object doesn't appear on print bed

Gravydog, 2019-12-21:
PrusaSlicer  2.1.1
win7 64bit
Prusa i3 MK3s

I open Prusaslicer and import an stl file. The object appears on the virtual print bed on the plater screen as expected. There are toolbars of icons on the left and top of the plater window. I set scale factors for the object successfully, and then I slice it. I export the G-code successfully. I highlight the stl file on the right and then delete it using "edit-delete selected" because the toolbars mentioned are no longer present. The filename is removed from the panel on the right. I choose Import and load another stl file, its name appears in the panel on the right but no object appears on the virtual print bed and no toolbars on the left or top. I can set the scale factors, change the view, etc. and no object ever appears on the print bed. If I click slice now, then the object appears in the scale I have set and the slice proceeds. The only way to get the toolbars back and make the object appear as soon as it's imported is to exit and restart Prusaslicer. I can't find any settings that I think could be involved. I am new, sorry if this is something I should know.

ps: is the lock next to scale factor new in 2.1.1? I remember scaling unequally before without noticing the lock or unlocking it.

Rob

lukasmatena, 2019-12-21:
[image]

The view automatically switches to GCode preview when you click Slice Now (and vice versa, switching to GCode preview initializes slicing).

The lock is there for a long time. If you use the scaling gizmo (left panel), you can scale non-uniformly without unlocking anything.

Gravydog, 2019-12-21:
Thanks, I see now. Clicking on the solid cube at the lower left brings everything to the state I see when initially opening the program and importing a file. I looked at the scaling gizmo and I'm sure that wasn't the method I used before to scale. Maybe I unlocked the lock and just don't remember.

Rob

closed 2019-12-21 by Gravydog