prusa3d/PrusaSlicer #3342: [PrusaSlicer 2.2.0-alpha1] Error opening PrusaSlicer -Internal error: Could not load bitmap: white/add

salmander, 2019-12-21:
### Version
PrusaSlicer 2.2.0-alpha1

OSX 10.15.2

Error log:

```
11:43:22: can't open file '/Applications/Prusa Beta/PrusaSlicer.app/Contents/MacOS/../Resources/icons/white/add.png' (error 2: No such file or directory)
11:43:22: Failed to load image from file "/Applications/Prusa Beta/PrusaSlicer.app/Contents/MacOS/../Resources/icons/white/add.png".
11:43:22: Internal error: Could not load bitmap: white/add

```

salmander, 2019-12-21:
Upon inspecting the contents directory, there are no PNGs available. All images are SVGs. Did something get missed before publishing this update? 

I did try to convert all *.svgs to png format and move to "white" directory. But, now I am getting the following error:
```
12:22:50: can't open file '/Applications/PrusaBeta/PrusaSlicer.app/Contents/MacOS/../Resources/icons/white/mode_simple_sq.png' (error 2: No such file or directory)
12:22:50: Failed to load image from file "/Applications/PrusaBeta/PrusaSlicer.app/Contents/MacOS/../Resources/icons/white/mode_simple_sq.png".
12:22:50: Internal error: Could not load bitmap: white/mode_simple_sq
```

I can't find `mode_simple_sq.svg` file anywhere so can't proceed.

I am really excited about this release. Hopefully, this will get fixed soon.

Thanks Prusa dev team.

Shama71, 2019-12-21:
I have same error message for add.png.

salmander, 2019-12-21:
I managed to fix the error by copying everything from the 2.1.1 version to the `icons/white` directory.
I was then able to launch PrusaSlicer.  By default, there were no profiles imported. Upon running configuration wizard it was constantly crashing. I changed my OS theme to light theme and voila. After restarting the app, I was able to go through the configuration wizard successfully.

lukasmatena, 2019-12-21:
You are correct, this is dark mode specific issue. It will be fixed soon.

eiswuerfel, 2019-12-21:
> I managed to fix the error by copying everything from the 2.1.1 version to the `icons/white` directory.
> I was then able to launch PrusaSlicer. By default, there were no profiles imported. Upon [...]

Thanks, worked for me on Manjaro Linux too.

bubnikv, 2019-12-22:
This issue will be fixed with PrusaSlicer 2.2.0-alpha2. Closing.

closed 2019-12-22 by bubnikv