prusa3d/PrusaSlicer #3343: Export adds .tmp to the file.

buliwyf42, 2019-12-21:
### Version
PrusaSlicer 2.2.0-alpha1

MacOs 10.15.3 Beta (19D49f)

When i try to export gcode it's saved with .gcode.tmp file extension.

cosmith71, 2019-12-21:
I see this issue too.

Windows 10.

lukasmatena, 2019-12-21:
Thanks, we are working on a fix.

DrLex0, 2019-12-21:
This breaks post-processing scripts: “PrusaSlicer has encountered an error” – “Post-processor can't find exported gcode file”

Area5142, 2019-12-21:
Same issue in Linux Mint 19.3 (AppImage).

italocjs, 2019-12-22:
Same here (saving .tmp), win10 1908

removing the .tmp shows that the gcode was exported correctly,

lukasmatena, 2019-12-22:
This is a bug that anyone not exporting to SD card will see. We already have a fix, it will be part of the next alpha (which I assume will be released tomorrow).

bubnikv, 2019-12-22:
It will be fixed in PrusaSlicer 2.2.0-alpha2. Closing.

closed 2019-12-22 by bubnikv