prusa3d/PrusaSlicer #3344: Uploading to octoprint fails

buliwyf42, 2019-12-21:
### Version
PrusaSlicer 2.2.0-alpha1

MacOs 10.15.3 Beta (19D49f)

When i try to upload to octoprint i get the attached error.

Perhaps i has something to do with my other issue where the exported file has the extension tmp. 

[image]

rmoravcik, 2019-12-21:
Same here, I'm using self compiled version from git on Linux

eiswuerfel, 2019-12-21:
Same error with Windows 10

pointbrink, 2019-12-21:
i have both issues as well.. exported gcode has the temp extension and i have the same http 411 error trying to go straight to octoprint.

lukasmatena, 2019-12-22:
Can you please retest with recently released alpha2? Thanks.

buliwyf42, 2019-12-22:
Already tested, and it works. :)

lukasmatena, 2019-12-22:
Great, thanks.

closed 2019-12-22 by lukasmatena