prusa3d/PrusaSlicer #3347: System Filament Profiles are ONLY compatible with System Print Profiles

neophyl, 2019-12-21:
### Version
_Version of PrusaSlicer used goes here_
2.2.0 Alpha1

### Operating system type + version
Windows 10 64 bit (but I dont think thats a factor)

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Not sure if this is a bug one of the known issues or just something I'm missing on my part but I was trying 2.2.0.  I have a default Mk3 printer added (added it again via the wizard too just to make sure it was in correctly).  I then added several of the filament options to see whats there now.  I added esun as its a filament I have and wanted to see what the profile was like compared to the one I normally use.
However I had Pretty PLA 4 selected as a print profile and NONE of the system filament profiles show up unless I also select a system Print profile.  The system Esun filament profile has Compatible Printers =All and Compatible Print Profiles = All ticked (its on its default settings) so why is it NOT compatible with the Pretty PLA  user profile ?  I cant see a single thing in the print profile that would bar it from matching ?

Personally I think adding the new filament profiles is great but NOT if you cant actually select them to use with ANYTHING but a stock profile.  What do I need to set to actually be able to use them ?

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Any

bubnikv, 2019-12-22:
What is "Pretty PLA user profile"? Print, filament or printer profile? Where did you get it?
It always helps, if you can attach as much information to the issue as possible including files.
For this issue, a zipped content of your PrusaSlicer-alpha directory would help.

I am currently not sure what the problem is. We made the system Filament profiles to be suppressed for non-system print or printer profiles. If your print or printer profiles are derived from our profiles, then everything shall work. We have renamed our profiles (see the note on logical names and the "@" sign in the change log), so as of now you will need to edit your user profiles and add the @ character into the names of the "inherits" field. Then everything shall work.

We may want to revise this tightened "compatibility" restriction on system profiles, but it is getting more complicated with the introduction of 3rd party configurations and the logical names of the profiles, and over-complicated rules are likely worse than a bit tighter, but clear rules.

neophyl, 2019-12-22:
The profile was downloaded from https://prusaslicerprofiles.com/download/pretty-pla-v4-0/.  Although I was trying to only use the print part of it and used the default mk3 printer added by the wizard and the esun filament also from the wizard.

My point is that locking things down in simple or advanced is fine but in advanced when the profile has compatibility = ALL checked for both Printers and Print Profile that it SHOULD be available for ALL printers and print profiles not JUST system ones.  I even unticked all and manually ticked every profile in the list including all of the non system ones as a test and it still didn't show up.

Heres a export of my config bundle-
[PrusaSlicer_config_bundle_2019_12_22.zip](https://github.com/prusa3d/PrusaSlicer/files/3992394/PrusaSlicer_config_bundle_2019_12_22.zip)

After looking at it some more after your reply I think it may be the new line under compatible printers profile of "nozzle_diameter[0]!=0.6 and printer_model!="MK2SMM" and printer_model!="MINI" and ! (printer_notes=~/.*PRINTER_VENDOR_PRUSA3D.*/ and printer_notes=~/.*PRINTER_MODEL_MK(2.5|3).*/ and single_extruder_multi_material)" that you have on all the new ones.  The and single_extruder_multi_material section.  However I cant see where that is defined in the printer profile when compared to the mk3 added by the wizard and the user defined ones that I have in there, most of which were derived from a mk3 profile added back in earlier versions.  Theres still only references to vendor and model in the notes so where is the single extruder multi material match coming from ?

bubnikv, 2019-12-22:
Please post a zipped content of your PrusaSlicer-alpha directory if it is not a secret. It contains your complete configuration. Also write down the print, filament and printer profile names you want to have selected at the same time. This will make it easiest for me to investigate your particular setup.

neophyl, 2019-12-22:
[PrusaSlicer-alpha.zip](https://github.com/prusa3d/PrusaSlicer/files/3992558/PrusaSlicer-alpha.zip)

Alpha folder attached.  Hopefully its the correct one (from Appdata>Roaming on win 10)

I was trying with several different printer profiles but I still have issue even when using printer Original Prusa i3 Mk3 , the filament profile is the Esun PLA profile and the Print profile is any of the user ones, although in this case I was trying Pretty PLA v4.  As I say if you select any print profile or printer profile that is NOT on of the system ones then the system filament profiles are not available.

Anything else I can do to figure out what I'm missing or doing wrong just ask.  Thanks for your help.

flinke-flasche, 2019-12-22:
I have the same problem with my own filament profiles. I can‘t select them with my print profiles. 

How can I fix this?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

neophyl, 2019-12-23:
I was experimenting with this some more and if I edit one of my Print profiles to include the @MK3 in the name AND have a valid inherits line copied from one of the system print profiles then it will show the System list of filaments for that profile once edited/restarted. 

One other thing of note is the new matching implemented now seems to also ignore the configuration value Show Incompatible Print and Filament Presets.  The System ones arent shown regardless of this setting.

bubnikv, 2019-12-23:
@neophyl Thanks for heads up.

> I was experimenting with this some more and if I edit one of my Print profiles to include the @mk3 in the name

This should not be needed.

> AND have a valid inherits line copied from one of the system print profiles then it will show the System list of filaments for that profile once edited/restarted.

This is required. We will fix that by introducing "renamed from" list of previous names of our system profiles.

> One other thing of note is the new matching implemented now seems to also ignore the configuration value Show Incompatible Print and Filament Presets. The System ones arent shown regardless of this setting.

Thanks, we will look into it after Christmas.

skellatore, 2019-12-25:
Is there an easy work-a-round for this? none of my filament profiles are showing in 2.2 :(

Pun-e, 2020-01-22:
I got this to work by making sure my custom print profile "inherits" from a stock one
Eg:
[image]

neophyl, 2020-01-22:
Did you manually edit any custom ones you already had before the 2.2.0 update and if so what exactly did you change to make them appear ?

Pun-e, 2020-01-22:
Find your custom print profile here:
[image]

Then add something here:
[image]

bubnikv, 2020-01-23:
The inheritance issue has been solved by implementing "renamed_from" profile attributes by 0d9022c5f6bc05154819617a72fae884776eada0 :

    Implemented "renamed_from" attribute of a system profile,
    so that references from user profiles and .3mfs / .amfs
    to system profiles are not being lost.
    
    If a system profile has no "renamed_from" profile name assigned,
    and the system profile name contains "@", then a profile name
    with the "@" is implicitely assumed to be the name, from which this
    profile has been renamed.

It will be part of the next 2.2.0-alpha.
Closing.

closed 2020-01-23 by bubnikv