prusa3d/PrusaSlicer #3351: GCODE file missing after exporting for postprocessing

tomvandeneede, 2019-12-21:
Version 2.2.0
Version 2.2.0-alpha1+-mac10.14
MACOS 10.15 Catalina

### Behavior
* Post script processor not working.   It seems the out creates a tmp file that is not renamed to the gcode file prior to starting the process to run the script.     The post script fails by PS2.2.0 saying the output GCODE file could not be found.   The actual script is never started.  Slicer reports  following error:         Post-Processor can't find exported gcode file

* I tested output file to P2PP 
* Post processing script should start and should process the script, PS2.2.0 stops before that point
* when the gcode file already exists, the post-processor runs on the previous file, not on the newly exported one. so it appears the rename/copy function from .tmp to final name fails to execute correctly

lukasmatena, 2019-12-21:
Duplicate of #3343, the fix is on the way. Closing.

closed 2019-12-21 by lukasmatena

tomvandeneede, 2019-12-21:
Appologies, failed to check before submitting.
Thanks for the prompt reply