prusa3d/PrusaSlicer #3353: Uploading to Flash Air card fails with "unknown error"

telepath, 2019-12-21:
### Version
2.2.0-alpha1

### Operating system type + version
debian stretch

### Behavior
* Uploading to Flash Air card fails with "unknown error"
* _Steps needed to reproduce the problem_
1. configure Flash Air: 
```
[Vendor]
UPLOAD=1
UPDIR=/    
WEBDAV=2
APPMODE=5
APPSSID=WiFi
APPNETWORKKEY=***********
VERSION=F15DBW3BW4.00.00
CID=02544d535731364754d1d45f7b011901
PRODUCT=FlashAir
VENDOR=TOSHIBA
DHCP_Enabled=NO
IP_Address=192.168.2.44
Subnet_Mask=255.255.255.0
Default_Gateway=192.168.2.1
Preferred_DNS_Server=192.168.2.1
NOISE_CANCEL=2
DNSMODE=1
appname=prusasd
MASTERCODE=redacted
LOCK=1
APPINFO=0000000000000000
APPAUTOTIME=300000
```
2. configure PrusaSlicer for Flash Air Upload:
[image]
3. Press "Test"
4. Message should say "Connection to FlashAir works correctly and upload is enabled
[image]
5. Slice a part and press the upload button
* _Expected Results_
  * File should be uploaded and appear on the FlashAir card
* _Actual Results_
  * Error message says "PrusaSlicer has encountered an error" and "Error uploading to print host: Unknown error occured."
[image]
  * terminal output says "[2019-12-21 20:39:36.993451] [0x00007f5df9848700] [error]   FlashAir: Request completed but no SUCCESS message was received."
  * File does not appear on FlashAir card

_Is this a new feature request?_
No.

bubnikv, 2019-12-22:
I wonder whether this issue is caused by https://github.com/prusa3d/PrusaSlicer/issues/3343 ?
Please retest with the upcoming PrusaSlicer 2.2.0-alpha2 release.

telepath, 2019-12-22:
Tested with alpha2, same result.

vintagepc, 2019-12-22:
I am observint the same issue with Alpha1, W-04 flashAir card on OpenSuSE.

jmd73, 2019-12-23:
Just chiming in. I just tested building from current master (version_2.2.0-alpha2, 220e2dbfec58cbbcff548d8211a17f0dcbe32aab), and it works on Windows 10 and expected.

I.e. Test is successful, and sending of object works too.

bubnikv, 2019-12-23:
So it seems that sending to FlashAir is not working on Linux. We will fix it after Christmas.

vintagepc, 2019-12-26:
I just tried out 2.2.0 alpha, and it appears to work for me. 

I reviewed the pull request and something about the URL formatting made me try prefixing the host with HTTP:// this time (something the original reporter also did not have). Success! The file uploads, but only appears in the root directory regardless of whether I prefix a folder in the name or not.

telepath, 2019-12-26:
confirmed, uploads works if the URL including HTTP:// is configured, but not if it's only the hostname.
@vintagepc thanks for testing this!

vintagepc, 2019-12-26:
It's weird because it *should* handle it and prefix the http:// if not present. 
Does specifying a directory work for you or does it also only dump the gcode on the root?

telepath, 2019-12-26:
Correction:
the upload starts now, but after it reaches 100%, the same error occurs and the file does not show up.

FidelCapo, 2020-01-07:
By what source did you configure the card? I use the W-03 version with the following configuration for testing and uploading is successful. You can try using this configuration, but I recommend backing up your SD card configuration.

[Vendor]
APPSSID=Internet
APPNETWORKKEY=**********
CID=[YOUR CID]
VERSION=[YOUR VERSION]
APPMODE=5
APPNAME=3D_printer
LOCK=1
PRODUCT=FlashAir
STA_RETRY_CT=0
UPDIR=/G-Code
UPLOAD=1
VENDOR=TOSHIBA
WEBDAV=2

vintagepc, 2020-01-07:
Is your question directed at me or telepath?

FidelCapo, 2020-01-07:
Sorry. It is for both of you. The more information, the better.

vintagepc, 2020-01-07:
[Vendor]
ID=PrusaMk3
CIPATH=/DCIM/100__TSB/FA000001.JPG
APPSSID=<snip>
APPMODE=5
APPNETWORKKEY=<snip>
VERSION=F15DBW3BW4.00.03
CID=<snip>
APPNAME=PrusaMk3
WLANAPMODE=0x82
UPDIR=/
LOCK=1
STA_RETRY_CT=0
WEBDAV=2
UPLOAD=1
PRODUCT=FlashAir
VENDOR=TOSHIBA

FidelCapo, 2020-01-07:
> [Vendor]
> ID=PrusaMk3
> CIPATH=/DCIM/100__TSB/FA000001.JPG
> APPSSID=[snip]
> APPMODE=5
> APPNETWORKKEY=[snip]
> VERSION=F15DBW3BW4.00.03
> CID=[snip]
> APPNAME=PrusaMk3
> WLANAPMODE=0x82
> UPDIR=/
> LOCK=1
> STA_RETRY_CT=0
> WEBDAV=2
> UPLOAD=1
> PRODUCT=FlashAir
> VENDOR=TOSHIBA

And this doesn't work?

vintagepc, 2020-01-07:
It works somewhat if I make sure the hostname is prefixed with http://, but only so far as that it doesn't upload anywhere but the root of the card, regardless of any paths I specify in the upload prompt. I've tried folder/file.gcode, /folder/file.gcode variants without success, it always just ends up in /file.gcode

bubnikv, 2020-01-09:
@telepath 
>Correction:the upload starts now, but after it reaches 100%, the same error occurs and the file does not show up.

What version of Toshiba FlashAir device are you using? W-2, W-3 or W-4? Is it still not working?

@vintagepc
> t doesn't upload anywhere but the root of the card, regardless of any paths I specify in the upload prompt
I suppose that is what the FlashAir's configuration word
UPDIR=/
does. We don't have source code of the FlashAir firmware, we can only guess.

vintagepc, 2020-01-09:
The web interface can upload to any arbitrary directory, so it must be possible. 
If we can't do it for flashAir, then the upload prompt dialog box shouldn't indicate you can specify folders in your path, and instead indicate the files will be uploaded to your "UPDIR" setting. 

Sounds like you may not have explicitly implemented any path upload support? I'll take a peek tonight and see what I can suss out. You also  appear to be interacting with the upload.cgi web service/interface on the flashAir; I'd suggest this support is less likely to break and be more maintainable if you instead interact with it over WebDAV. Then, in theory, any webDAV compatible remote host (not just a FlashAir card) could be used.

https://en.wikipedia.org/wiki/WebDAV

That is the protocol I use to access it from my Linux file manager.

bubnikv, 2020-01-10:
The FlashAir has been contributed to us. We don't have resources to invest
too much time into it.

> If we can't do it for flashAir, then the upload prompt dialog box
shouldn't indicate you can specify folders in your path, and instead
indicate the files will be uploaded to your "UPDIR" setting.

That may be a good solution as for now.

In regard to WebDAV, if it is contributed, we will merge it.

I am not sure whether the upload.cgi could support subdirectories, the
documentation is sparse and as stated above, we have little resources as of
now for the FlashAir upload implementation.

vintagepc, 2020-01-10:
I did a quick poke around for info on upload.cgi and found a useful tidbit here https://mattshub.com/blog/2017/04/11/flashair-sd-card
which has some improved web I/F files. I think we may just need to query upload.cgi with an UPDIR parameter before actually sending the file. 

I recognize that resources are focused elsewhere; I will see if I have some time to make changes/test this weekend and file a PR.

bubnikv, 2020-01-10:
I did a quick back of the envelope calculation yesterday.

Number of open issues, each issue requires 10 developer days. We are 6
developers, each working 5 days a week. When counting in the vacations, we
will clean up the github issue back log in 8 years.

vintagepc, 2020-01-10:
Oof... I'll keep that in mind if I'm on the hunt for a new job :P

vintagepc, 2020-01-11:
Good news, my cursory test appears successful. I'll polish the code a bit and file a PR for directory support. (The current code just ignores the directory completely)

vintagepc, 2020-01-11:
Pushed! :-)

lukasmatena, 2020-01-31:
Should be fixed after #3511 was merged - thanks @vintagepc .The fix will be part of the next release (2.2.0-alpha3).
Closing the issue.

closed 2020-01-31 by lukasmatena