prusa3d/PrusaSlicer #3355: Export to other location

evolution80, 2019-12-21:
### Version
PrusaSlicer 2.2.0 Alpha1

### Operating system type + version
Windows 10 Home 64bit

### 3D printer brand / version + firmware version (if known)
MK3s MMU2s Stock

### Behavior
Export to external SD works as expected, however; I'm not able to export to 2nd internal hard drive
Repeatable everytime. I navigate to folder I want to export / save gcode - file doesn't save

lukasmatena, 2019-12-21:
Seems like a duplicate of #3343. I would bet the file is actually saved, but with an extra .tmp extension, is that correct?

evolution80, 2019-12-21:
They are, I should've checked further *.tmp files don't generally show on dialogue upload boxes when deciding to upload to Octoprint. 
Thanks for reply.
Can be closed as believe part of 3343 issue.

closed 2019-12-21 by evolution80