prusa3d/PrusaSlicer #3357: [PrusaSlicer v2.2.0-alpha1] Center of rotation is not where it should be

mgg4, 2019-12-21:
### Version
PrusaSlicer v2.2.0-alpha1

### Operating system type + version
Windows 10 Build 1909

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3/MMU2

### Behavior
Rotation origin is not in the center of the build platform image.

When rotating the build plate image, the center of rotation is not located in the center of build platform.  It is actually not even located in the build platform image at all, but appears to be to the front/left of the visible origin arrows.

This behavior occurs while rotating using a normal mouse (click/drag), and using the SpaceMouse Wireless.

This is not a feature request, but it is a new issue.

#### Project File (.3MF) where problem occurs
N/A

Panayiotis-git, 2019-12-21:
I do not have a SpaceMouse, but the rotation center is at the center of the plate, unless I have already zoomed to another object

mgg4, 2019-12-21:
I have done a bit more exploring.  It appears that if the build platform is panned or zoomed, the rotation origin stays fixed in space, rather than moving with the platform.  I should have done a bit more research before opening this issue.  Please advise if you would prefer I amend this ticket, or open a new ticket with a better description.

Panayiotis-git, 2019-12-21:
I understand now what you are reporting and you are right. I don't know if this is the intended behavior but I can reproduce it at 2.1.0 also.

bubnikv, 2019-12-22:
The problem with rotation center for rotation in 3D is in that "where it should be". Where should it be?

One either makes the center of rotation explicit (show it in the scene, allow one to modify it explicitely), or one makes it hidden and one tries to make the application to select an intuitive
 center of rotation. We chose the later, and the center of rotation is set automatically and it is only moved left / right / up / down when panning.

bubnikv, 2020-03-15:
Center of rotation may be moved to the center of object with the "B" hot key (center to bed) or "Z" (center to object). We recommend you to map these hot keys to the buttons of the 3DConnexion device in the 3DConnexion driver.

Closing.

closed 2020-03-15 by bubnikv