prusa3d/PrusaSlicer #3359: [PrusaSlicer v2.2.0-alpha1] Ender-3 picture not shown in AppImage

probonopd, 2019-12-21:
In the release notes, a picture for the Ender 3 is shown:

[image]

This picture is not shown in the AppImage:

[image]

The build plate is shown correctly though:

[image]

And the Prusa mini picture is also shown correctly:

[image]

bubnikv, 2019-12-22:
Thanks for heads up. It works on Windows and OSX. The image is in both the Linux tar.gz package and the AppImage. I have no idea why the image is not shown, maybe the png is not compatible with Linux libpng. I will leave this issue to be resolved by one of our in house Linux fans.

lukasmatena, 2020-01-06:
The problem was much simpler. Icon name had different case than profile name, so the icon was not found on Linux case-sensitive filesystem. Fixed with https://github.com/prusa3d/PrusaSlicer/commit/b08a315a469c38f23a30908e2cf6170c11e90ad1, it should work in the next release. I also added a "NO IMAGE" icon to show when this happens.

Thanks for reporting this. Closing.

closed 2020-01-06 by lukasmatena