prusa3d/PrusaSlicer #3360: issues compiling with VS2019

italocjs, 2019-12-22:
### Version
2.2.0-alpha1

### Operating system type + version
Windows 10 1903

### Behavior
I 99% sure this is me doing something wrong.  I want to understand better how the program works in order to be more helpful and create some functions, one of the first steps i'm trying is to learn how to compile the source (never did anything similar before), so i followed the [guide](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md)  but did not suceeded, i never used visual studio before (my programming skills are arduino, lol).

The error i'm having when trying to compile is posted here  [pastebin
](https://pastebin.com/FxFMLfR2).

i also read something about compiling libraries, i have no idea which ones are, and where to download them.  there is nothing on the guide.

italocjs, 2019-12-22:
Well, ended up discovering that i needed to install boost 1.66,  then intel tbb, but still nothing works, there is no documentation saying to install these, i ended up finding stuff on the error reports,  we really need a better documentation for how to compile this, if i'm lucky enough to make it work i'll write one.

lukasmatena, 2019-12-22:
The instructions [here](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md#building-the-dependencies-package-yourself) that you claim to have followed are outdated, but most of it still holds. The description of your problems with TBB etc. suggests that you did not do what the paragraph about dependencies says. I think there is no lacking documentation here.

italocjs, 2019-12-22:
Hi @lukasmatena , thanks for the answer, i tried building the libraries but also returned erros (the one attached originally on the pastebin).

any ideas?

italocjs, 2019-12-22:
This is what i get when i try to build the dependencies (not sure if i should do it, but i downloaded and installed boost and tbb because it was saying it was no found before)
https://pastebin.com/1mfV1J6S

This is what is see on the output of the visual studio cmake (it opens automatically)
https://pastebin.com/ENknkynf

douggorgen, 2019-12-23:
I am now able to compile and run PrusaSlicer using Visual Studio 2019. Hope this helps.

Notes:

1) I am a total noob, so go easy on me. This procedure is probably not optimal.

2) I installed all of Visual Studio because I was getting errors for missing includes. The errors went away after installing everything. If anyone knows the exact minimum set of options, please post here. For me, it's not worth my time to figure out the minimum set of options just to save a few GB.

3) After this worked the first time, I uninstalled everything and did it over to make sure it still worked.

4) I have not tried all the combinations of builds i.e. debug, release, 32 bit, 64bit, cross compile, etc. So far it works for me in Windows 64 bit debug mode.

5) I did this on my gaming machine which had never before been used for code development so there should not have been anything present that helped me without my knowing about it.

6) These build steps are of course based on the "Building PrusaSlicer on Microsoft Windows" guide. I repeat those steps to show the actual parameters that I used.

Now for the procedure I followed:

1) Install Visual Studio Community 2019 from
	visualstudio.microsoft.com/vs/
	Select all workload options	// I don't know which ones are required, so I selected them all
			
2) Install git for Windows from
	gitforwindows.org
	download and run the exe accepting all defaults

3) download PrusaSlicer-master.zip from github
   I downloaded this to d:\downloads\PrusaSlicer and unzipped to d:\downloads\PrusaSlicer\PrusaSlicer-master\ so this will be my prefix for all my steps. Substitute your prefix.

3) go to the Windows Start Menu and Click on "Visual Studio 2019"->"x64 Native Tools Command Prompt" to open a command window

4) cd d:\downloads\PrusaSlicer\PrusaSlicer-master\deps

5) mkdir build

6) cd build

7) cmake .. -G "Visual Studio 16 2019" -DDESTDIR="d:\downloads\PrusaSlicer\PrusaSlicer-master\"

8) msbuild /m ALL_BUILD.vcxproj   // This took 13.5 minutes on my machine: core I7-7700K @ 4.2Ghz with 32GB main memory

9) cd d:\downloads\PrusaSlicer\PrusaSlicer-master\

10) mkdir build

11) cd build

12) cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="d:\downloads\PrusaSlicer\PrusaSlicer-master\usr\local"

13) open Visual Studio for c++ development (VS asks this the first time you start it)

14) Open->Project/Solution or File->Open->Project/Solution (depending on which dialog comes up first)

15) click on d:\downloads\PrusaSlicer\PrusaSlicer-master\build\PrusaSlicer.sln

16) Debug->Start Debugging or Debug->Start Without debugging
    PrusaSlicer should start. You're up and running!

italocjs, 2019-12-23:
Thank you so much @douggorgen , i will try doing as you said!

italocjs, 2019-12-23:
@douggorgen it worked perfectly, i'll suggest a change on the "how to compile" from the official post, can i use your instructions?

douggorgen, 2019-12-23:
> @douggorgen it worked perfectly, i'll suggest a change on the "how to compile" from the official post, can i use your instructions?

I'm glad it worked for you.  Feel free to use my instructions, however they are not complete because they do not address 32 bit and Linux development. I have no plans to work in either of those environments.

closed 2019-12-25 by italocjs