prusa3d/PrusaSlicer #3364: Any value infill will fill 100% small object

tonycstech, 2019-12-22:
10x10x100 cuve any value in infill parameter fills entire object.
Larger object placed next to it gets proper infill.
[image]

lukasmatena, 2019-12-22:
The picture looks that the "100% infill" is actually solid infill near the top. Turn it off and see (Print Settings->Layers And Perimeters->Horizontal Shells->Top/Bottom Solid layers - set both to zero).

tonycstech, 2019-12-22:
I dont want to disable solid bottom and top, i need them solid.
What i am saying is infill is not working for small 10x10 cube and fills in solid all the way up regardless of settings, unless settings is 0%, then it prints hollow, but i dont want hollow, i want some infill.

lukasmatena, 2019-12-22:
Could you post a 3MF?

ieRev, 2019-12-22:
In the Expert options, Print Settings, Infill.. Check the Solid infill threshold area value.

Objects smaller than that value will be printed solid regardless of the infill value

tonycstech, 2019-12-22:
Solid infill threshold 0 did the trick.
Should have been 0 by defauilt. This behavior  is unexpected. No slicer behaves like that so i didnt expect it here either.

bubnikv, 2020-01-02:
The full infill makes a lot of sense to reinforce thin parts, you may ask my kids how easy it is to break pony's tail. What is missing is the configurability of the maximum area to be filled in.

closed 2026-09-29 by prusaslicer-bot[bot]