prusa3d/PrusaSlicer #3365: Support Part Marker to Cancel parts one-by-one with OctoPrint

chatelao, 2019-12-22:
### Version
_Version 2.1.0

### Behavior
* Cannot cancel parts, as the names are not included in GCode
* _Steps 
  * Install https://plugins.octoprint.org/plugins/cancelobject/
* _Expected Results_
  * Show one name per part
* _Actual Results_
  * No part names are exported

This may be a feature request.

lukasmatena, 2019-12-22:
Print Settings->Output Options->Label objects

chatelao, 2019-12-23:
Thanks for the answer. Is there any reason, this option is not active by default?

bubnikv, 2020-01-02:
> Is there any reason, this option is not active by default?

Most of our customers are printing without OctoPrint.

Could you please close this issue?

chatelao, 2020-01-03:
Sure, you have the numbers.

closed 2020-01-03 by chatelao