prusa3d/PrusaSlicer #3366: Error on change Infill to 100%

coquelet, 2019-12-22:
### Version
2.2.0-alpha1-win64

### Operating system type + version
Windows 10 1903 18362.535 64bits
### 3D printer brand / version + firmware version (if known)
_MK3S - 3.8.1-2869

### Behavior
When changing infill parameter to 100%, slicer close after this error message :
Unhandled unknow exception;terminating the application.
After cklicking OK, message is :
08:54:43: Internal error: boost::bad_format_string: format-string is ill-formed
08:54:43: Internal error: boost::bad_format_string: format-string is ill-formed

lukasmatena, 2019-12-22:
This is actually a bug in French localization - unescaped % character in the localized string. We will fix that, thanks. @YuSanka @bubnikv

lukasmatena, 2019-12-22:
Actually, Italian and Spanish appear to suffer from the same...

bubnikv, 2019-12-22:
and traditional Chinese (Thaiwan) as well.

bubnikv, 2019-12-22:
and Ukrainian

bubnikv, 2019-12-22:
It will be fixed in PrusaSlicer 2.2.0-alpha2. Closing.

closed 2019-12-22 by bubnikv