prusa3d/PrusaSlicer #3368: [PrusaSlicer v2.2.0-alpha1] Error on switching language from German to English on Linux

th33xitus, 2019-12-22:
### Version
Version:   2.2.0-alpha1+linux-x64
Build:     PrusaSlicer-2.2.0-alpha1+linux-x64-201912201649

### Operating system type + version
Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.4.2-1-MANJARO x86_64

### Behavior
I always use my slicers on english, even though i am a native german speaker.
Since it has been quite a while that i tried a new alpha-version of PrusaSlicer i didn't had any alpha-profiles on my machine anymore. So PrusaSlicer started in german.
I tried to switch language to english but the following message popped up:
```
Switching PrusaSlicer to language en_GB failed.
You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands.
```

I use PrusaSlicer 2.1.0 on my machine on english as well, therefore i don't quite understand why i can't switch the translation and get the error message. Besides the german locales (obviously) i have the `en_GB` and `en_US` locales installed on my machine. So that error makes not a lot of sense to me. 

Another reason why it does not make a lot of sense is, that i can open up the newly generated `PrusaSlicer.ini` of PrusaSlicer 2.2.0-alpha1 and simply paste the following line which i took out of the `PrusaSlicer.ini` generated by my PrusaSlicer 2.1.0 Version:
`translation_language = en_US`
Saving the file and restarting PrusaSlicer 2.2.0-alpha1 now gives me the english interface.

Then i tried a second thing to make sure it is still no problem with my machine.
I created a `.desktop` file with the `--datadir` variable included and set up a new directory for the configuration files of the program. The program started in german again.
Now, just for fun i pasted the `translation_language = en_US` line into the newly created `PrusaSlicer.ini` which is located in the directory given by the `--datadir` parameter and deleted the line from the `PrusaSlicer.ini` file in the `.PrusaSlicer-alpha` directory in my users home-folder.
I can now start PrusaSlicer 2.2.0-alpha1 with the english interface by using the `.desktop` file and start it with the german interface by simply executing the AppImage itself.
[image]
Started with `.desktop` file on the left, started with the AppImage itself on the right.

Same goes for the non-AppImage versions. So it is not a problem only related to the AppImage.

Using `translation_language = en_GB` instead of `translation_language = en_US` results in the same error message than before. So that's another point which makes no sense to me. Why is there no error message appearing if `en_US` is used if the program expects the `en_GB` locale to be installed when changing the UI to english?

th33xitus, 2019-12-22:
Short update:
I fiddled around a little bit more and now i am not 100% sure anymore that it is not a problem with my machine lol. Therefore i close the bug-report here and might reopen it if i am 100% sure again that there is no issue on MY side.

closed 2019-12-22 by th33xitus

salfter, 2020-12-01:
This same error just caught me out on 2.3.0-beta1 (running the 64-bit AppImage on Gentoo).  Adding ```translation_language = en_US``` (which is the correct locale for me) to PrusaSlicer.ini got it running.  Rerunning locale-gen did nothing useful, and dpkg-whatever isn't a valid command on Gentoo...distro-specific commands probably shouldn't be part of error messages in AppImages. 

Just for sh*ts and grins, I tried changing the translation_language setting away from en_US.  Neither en_GB (what PrusaSlicer is apparently requesting) nor de_DE worked.  It's almost like it says it's asking for en_GB, but in reality is actually trying to use en_US, or something.

bubnikv, 2020-12-01:
Fixed with 2e73a759516071b201e6de6b8329fa4d3cbfab43

    Linux specific fix for switching languages: PrusaSlicer 2.3 forces
    en_GB locale when switching to English from the user interface,
    while often Linux users have just en_US locales configured.
    
    With this commit the user will have to select English (U.S.) or
    English (U.K.). Usually English (U.S.) will work.
    
    Vojtech is scared to touch the code for the other platforms (Windows and Mac)
    as the language switching has been reworked multiple times and it is quite
    fragile.

bubnikv, 2020-12-01:
Switching the language is really a mess to do it right in a multi platform fashion. I cannot count hours I have already spent on this. Each platform does it differently, Windows even declare C locales + Win32 locales, which are different animals. wxWidgets consider the default English to be en_GB.

For this new issue I did a very minimalistic change, which should make you Linux guys happy and hopefully I could sleep well also.

bubnikv, 2020-12-01:
> Adding translation_language = en_US (which is the correct locale for me)
to PrusaSlicer.ini got it running. Rerunning locale-gen did nothing useful,
and dpkg-whatever isn't a valid command on Gentoo...distro-specific
commands probably shouldn't be part of error messages in AppImages.

You would have to run
```
sudo locale-gen en_GB.UTF-8
```

to produce the U.K. locales that PrusaSlicer 2.3.0-beta1 requires when
switching to English.
PrusaSlicer 2.3.0-beta2 will offer you a choice of both English (U.S.) and
English (U.K.)
In the future we may try to detect which locales you have generated on your
Linux box, but I believe the current implementation is sufficient.

shiftingtech, 2020-12-05:
Canadian Checking in here:  I'm not a huge fan of "In the future we may try to detect which locales you have generated on your
Linux box, but I believe the current implementation is sufficient."  

Every other application on my system manages to function fine if en_CA.UTF-8 is my only available locale.  Sometimes my systems end up with a US or GB locale available sometimes not.  But it seems awkward that prusaslicer would be the only application which can't function on an en_CA only system.

bubnikv, 2020-12-06:
@shiftingtech 

I understand your worries. Now if the en_CA.UTF-8 is your active locale, then everything should work just fine, right?
The only problem you would have is if you had a different locale active (for example Spanish) and trying to switch the language to English from the PrusaSlicer UI.

So do you experience any practical issue with PrusaSlicer 2.3.0-beta2?

shiftingtech, 2020-12-07:
>I understand your worries. Now if the en_CA.UTF-8 is your active locale, then everything should work just fine, right?
>The only problem you would have is if you had a different locale active (for example Spanish) and trying to switch the language to English from the PrusaSlicer UI.

That would be fine.  That is not the behavior currently observed though.

>So do you experience any practical issue with PrusaSlicer 2.3.0-beta2?

Yes.  When I attempt to run it on a system that only has en_CA.UTF-8  available, it fails to start, after complaining about not being able to switch to en_GB

bubnikv, 2020-12-07:
> Yes. When I attempt to run it on a system that only has en_CA.UTF-8 available, it fails to start, after complaining about not being able to switch to en_GB

Will it start if you delete the translation_language key from 
.config/PrusaSlicer/PrusaSlicer.ini
?

BTW I would really appreciate if the Linux community solved these issues for us and we could concentrate on making the algorithms better.

shiftingtech, 2020-12-07:
Unfortunately, that key is not present in my config.

belese, 2020-12-10:
Hello, iv'e got similar problem, 
it ask for locale fr_FR, and doesn't start
i live in french belgium, and my locales was :
```
sudo locale-gen
Generating locales (this might take a while)...
  fr_BE.UTF-8... done
Generation complete.
```
i've deleting my Prusa profile folder but no luck.
i've tried with different translation_language in ini file (en_US or en_GB) but same error, ( don't remember if i've tried with 'fr_BE') but no luck.
my solution was to add fr_FR to my locale with 
```
sudo dpkg-reconfigure locales

now my locales are 

sudo locale-gen
Generating locales (this might take a while)...
  fr_BE.UTF-8... done
  fr_FR.UTF-8... done
Generation complete.

```
and now, it start.

bubnikv, 2020-12-12:
Please try with PrusaSlicer 2.3.0-beta3, @lukasmatena reworked the Linux specific code to align the locale with those installed on your Linux box.

belese, 2020-12-12:
Yes problem is fixed for me, 
i've removed fr_fr from my locale and beta3 start without error.
Thanks!