prusa3d/PrusaSlicer #3370: Perimeter modifier breaks perimeters (end result gives ridges)

Mirarkitty, 2019-12-22:
### Version
2.2.0-alpha

### Behavior
* load an object
* make a modifier
* set perimeters to a different value than the object
* perimeters now are local to the modifier/not modifier and "bends in" as if the modified part is a different object
* printing this gives an ugly ridge instead of a smooth surface and probably less strong part

(This is probably a hard problem to solve, but it currently does give ugly results.)

[image]

This modifier resulted in a weaker part and an ugly ridge.

inner-ari, 2020-12-16:
Question: since issue #2982, which refers to _infill_ modifiers affecting perimeters, has been solved, Is it possible to apply a similar solution here as well (and #3495 and #5490), to address _perimeter_ modifiers? I.e. to at least address the surface issue and maybe even the extraneous internal perimeters? (I can confirm that the problem still exists in PrusaSlicer-2.3.0-beta3, see #5490 .) 

And just to amplify the issue of structural weakness issue that @Mirarkitty (here) and I (#5490) pointed out. The artificial perimeters inside the body create a cleavage surface, which could definitely weaken thin parts -- for example:

[image]

The modifier mesh, which requests 3 perimeters instead of two :
[image]

Thank you!

tonysmartantzoulatos, 2022-06-02:
Has there been any update on this issue?

stepikovo, 2022-06-06:
@tonysmartantzoulatos not really, this is the current intended behaviour, a "workaround" is making the modifier smaller so it's not crossing the shell of the model. How modifiers work might be changed in the future though.

tonysmartantzoulatos, 2022-06-08:
> @tonysmartantzoulatos not really, this is the current intended behaviour, a "workaround" is making the modifier smaller so it's not crossing the shell of the model. How modifiers work might be changed in the future though.

For most parts we "like" to print the solution is not possible since we print with single 0.4 wall and we have minot parts on that wall that need to have more infill or perimeters. Hope it will be solved in the near future

Shushuda, 2022-07-16:
> @tonysmartantzoulatos not really, this is the current intended behaviour, a "workaround" is making the modifier smaller so it's not crossing the shell of the model. How modifiers work might be changed in the future though.

This workaround is impossible when the modifier is supposed to modify parameter speeds. It has to cross the shell of the model. It also cannot be made bigger, because it will affect the entire model on these layers, as apposed to just a small part of the model. For example, I want to print a figure with arms and I want the arms to print slower, but the rest of the model to print normally. The modifier will have to cross the perimeters in this case (between the arms and body) and there is no workaround.

Are there any plans to change this behaviour or is it more of a "might be changed, might be not"? Atm modifiers are useless for anything but infill or entire layers.

archanut, 2022-11-27:
@Shushuda  This workaround is impossible when the modifier is supposed to modify parameter speeds.

I agree and I believe this should be corrected to work without adding perimeters

BMacours, 2023-01-11:
+1 for not breaking perimeter especially when modifying parameters which are not affecting perimeters (i.e. speed, ...). I'm printing small technical parts with overhangs where I have to reduce speed (to limit curling) without changing the part visual aspect nor it's strength caused by added breaking perimeters. Thanks!

michalelektryk, 2023-10-23:
So there's currently no way to add more perimeters  to holes in the print without adding more perimeters to entire model or ring of perimeter inside?

nergen, 2023-11-01:
+1 to this issue, perimeter modifier become quite useless because of that

giuigiui, 2023-12-11:
I'm looking forward to fixing this. I'd like to slow down the print around slightly overhanging external perimeters. And of course I don't want to create unnecessary perimeters inside the print, since it would make the structure prone to breaking
[image]

gwicke, 2024-03-09:
Another +1. I stumbled into this issue while trying to add additional perimeters (and increased infill density) in highly stressed areas. For now, my work-around is limiting the reinforcement to infill only, but this is not sufficient for some use cases I would like to print. Cura supports adding perimeters in modifier regions more cleanly.

corndog2000, 2024-09-09:
Bump +1. Also having this issue.

hugo-id, 2024-10-25:
I'm also running into this issue.

I'm working on the following design:

With single perimeter:
[image]

With dual perimeters:
[image]

But due to #4767, I'm getting random tiny infills everywhere.

The solution to work around #4767, are modifiers, but when I try to use one, I get the following:
[image]

This print will fail immediately.

The default behaviour seems a lot more complicated to me than what expected behaviour should be, but maybe this is a holdover from slicers' fear of open-loop extrusions.

Hopefully this will get looked into soon as I'm absolutely beating my head against a wall here.

dmbaio, 2024-11-20:
FYI: [PrusaSlicer 2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1) fixes this, but it looks like it only prevents fuzzy skin settings on the modifier from triggering this, not other settings like perimeters.

suxscribe, 2025-05-01:
Same here. It adds 3 perimeters (below top infill) around the modifier box, while set to 1 perimeter. It doesn't cross any perimeters

Github fails to upload screenshot, so [https://imgur.com/YJhVr3p](url)

Buuuh, 2025-05-02:
+1 
still broken. 

I want to make a region of the print stronger by e.g. printing 5 instead of 3 perimeters. 
Making new perimeters along the modifiers borders results in the opposite, because there comes a new weakpoint up where the lines (don't) connect. 
The modifier shall just add perimeters where there are already perimeters. And especially let those who are already there being connected. 

Iirc when i was using Cura years ago, it already did this "as expected". So this is not "as it's intended to be", it's just how prusa slicer messes it up. 

Prusa does it like this:

[image]

What i at least expect is having the lower perimeter count connected:

[image]

I could live with the unneccessary inner perimiters being generated but breaking those outer perimeters is a bust.

MilesStarr, 2026-01-05:
+1

Trying to make a separate volume to resin fill that has thicker perimeters and a manually added 100% infill zone to partition what gets resin filled vs hollow, but instead the perimeter turns inward.

[image]

[image]

Shushuda, 2026-09-02:
This issue is still occurring in 3.0.0 alpha.