prusa3d/PrusaSlicer #3371: [PrusaSlicer v 2.2.0-alpha2] Crashed when checking wipe tower

salmander, 2019-12-22:
Steps to reproduce:
* Open slicer
* Load model
* Choose MMU profile
* Click on print settings
* Under `Multiple Extruder > Wipe Tower`, check `No Sparse Layers`
***Crash***

Log Below:

```

Process:               PrusaSlicer [34160]
Path:                  /Applications/PrusaSlicer Alpha.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.2.0-alpha2+-201912221330 (PrusaSlicer-2.2.0-alpha2+-201912221330)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [34160]
User ID:               501

Date/Time:             2019-12-22 21:30:33.450 +0000
OS Version:            Mac OS X 10.15.2 (19C57)
Report Version:        12
Bridge OS Version:     4.2 (17P2551)
Anonymous UUID:        230D2DC7-F3F9-FDD1-F896-FF1C63D52920

Sleep/Wake UUID:       BF1F1823-8636-47FD-B4A9-8F1F6E647F5F

Time Awake Since Boot: 61000 seconds
Time Since Wake:       990 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [34160]

VM Regions Near 0:
--> 
    __TEXT                 00000001009ae000-0000000101965000 [ 15.7M] r-x/r-x SM=COW  /Applications/PrusaSlicer Alpha.app/Contents/MacOS/PrusaSlicer

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff33671db0 CFDictionaryGetValue + 33
1   com.prusa3d.slic3r/           	0x0000000101203cef wxMacCoreGraphicsContext::DoDrawText(wxString const&, double, double) + 319
2   com.prusa3d.slic3r/           	0x000000010128aa93 wxGCDCImpl::DoDrawText(wxString const&, int, int) + 147
3   com.prusa3d.slic3r/           	0x0000000100e62686 Slic3r::GUI::GLCanvas3D::LegendTexture::generate(Slic3r::GCodePreviewData const&, std::__1::vector<float, std::__1::allocator<float> > const&, Slic3r::GUI::GLCanvas3D const&, bool) + 1414
4   com.prusa3d.slic3r/           	0x0000000100e74882 Slic3r::GUI::GLCanvas3D::load_preview(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<Slic3r::Model::CustomGCode, std::__1::allocator<Slic3r::Model::CustomGCode> > const&) + 434
5   com.prusa3d.slic3r/           	0x0000000100f47e24 Slic3r::GUI::Preview::load_print_as_fff(bool) + 1476
6   com.prusa3d.slic3r/           	0x0000000100f4876a Slic3r::GUI::Preview::reload_print(bool) + 74
7   com.prusa3d.slic3r/           	0x0000000100f8ab8e Slic3r::GUI::Plater::priv::update_background_process(bool, bool) + 286
8   com.prusa3d.slic3r/           	0x0000000100fc7295 wxEventFunctorFunctor<wxEventTypeTag<wxTimerEvent>, Slic3r::GUI::Plater::priv::priv(Slic3r::GUI::Plater*, Slic3r::GUI::MainFrame*)::$_19>::operator()(wxEvtHandler*, wxEvent&) + 37
9   com.prusa3d.slic3r/           	0x00000001014173ee wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
10  com.prusa3d.slic3r/           	0x000000010141715b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
11  com.prusa3d.slic3r/           	0x0000000101417070 wxEvtHandler::ProcessEvent(wxEvent&) + 96
12  com.prusa3d.slic3r/           	0x000000010141758c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
13  com.prusa3d.slic3r/           	0x00000001013e1f3a wxTimerImpl::SendEvent() + 90
14  com.apple.CoreFoundation      	0x00007fff33709804 __CFRUNLOOP_IS_CALLING_OUT_TO_A_TIMER_CALLBACK_FUNCTION__ + 20
15  com.apple.CoreFoundation      	0x00007fff337093be __CFRunLoopDoTimer + 859
16  com.apple.CoreFoundation      	0x00007fff33708e9e __CFRunLoopDoTimers + 317
17  com.apple.CoreFoundation      	0x00007fff336edaed __CFRunLoopRun + 2213
18  com.apple.CoreFoundation      	0x00007fff336ecbd3 CFRunLoopRunSpecific + 499
19  com.apple.HIToolbox           	0x00007fff3224365d RunCurrentEventLoopInMode + 292
20  com.apple.HIToolbox           	0x00007fff3224339d ReceiveNextEventCommon + 600
[... 592 lines cut ...]
===========                     =======  ======= 
TOTAL                              7.3G     3860 
TOTAL, minus reserved VM space     6.0G     3860 

Model: MacBookPro16,1, BootROM 1037.60.58.0.0 (iBridge: 17.16.12551.0.0,0), 8 processors, 8-Core Intel Core i9, 2.3 GHz, 32 GB, SMC 
Graphics: kHW_IntelUHDGraphics630Item, Intel UHD Graphics 630, spdisplays_builtin
Graphics: kHW_AMDRadeonPro5500MItem, AMD Radeon Pro 5500M, spdisplays_pcie_device, 8 GB
Memory Module: BANK 0/ChannelA-DIMM0, 16 GB, DDR4, 2667 MHz, SK Hynix, -
Memory Module: BANK 2/ChannelB-DIMM0, 16 GB, DDR4, 2667 MHz, SK Hynix, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Oct 26 2019 10:24:57 version 9.112.2.0.32.5.40 FWID 01-95c65950
Bluetooth: Version 7.0.2f4, 3 services, 18 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.1 Bus
USB Device: Apple T2 Bus
USB Device: Composite Device
USB Device: Touch Bar Backlight
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 51.7
Thunderbolt Bus: MacBook Pro, Apple Inc., 51.7
```

lukasmatena, 2019-12-23:
Thanks for the crashdump, but it makes little sense to me. It seems that it has little to do with the wipe tower, it crashes when it switches to Preview and generates legend textures. I therefore assume it crashes when slicing is initiated, not when you select the option, correct? Also, have you ever seen it crash without the wipe tower? Does it happen with 2.1.0? Do you run PrusaSlicer in English or localized?

I found many bug reports from many different projects of a crash in CFDictionaryGetValue on Mac. Don't know what to take from it. I am not sure if we can fix it without being able to reproduce the issue. Are you able to reproduce it consistently? Are you able to compile PrusaSlicer yourself so we could use you to experiment (if we had some ideas)? Thanks.

salmander, 2019-12-23:
I was able to reproduce it by doing the following:
* Open slicer
* Don't load any model, go straight to the `Print Setting > Multiple Extruders`.
* `Check` and then `Uncheck` `No Sparse Layers`
***Crash***

salmander, 2019-12-23:
I am happy to be a guinea pig. I have never tried to compile it myself but I am sure shouldn't be that hard. Feel free to send me branch/commit that you want me to checkout and try.

lukasmatena, 2019-12-23:
Thanks for the info. I have honestly no clue what is happening. We'll let you know if there's anything we would like you to try for us (unless we manage to reproduce it ourselves).

I guess it's MacOS (and maybe 10.15 Catalina) specific.

salmander, 2019-12-23:
It seems Catalina specific. I tried to follow the steps on my work macbook which is running Mojave and it didn't crash.

bubnikv, 2019-12-23:
> It seems Catalina specific. I tried to follow the steps on my work macbook which is running Mojave and it didn't crash.

Thanks for heads up. What a thrill. We will look into this issue after Christmas, we have a MacBook with Catalina to test on.

thisiskeithb, 2019-12-27:
Running 2.2.0-alpha2+ on Mojave here and I get repeatable crashes if I have a model loaded and try to edit any of the settings. Fresh settings still don't seem to help.

lukasmatena, 2020-01-02:
@thisiskeithb Thanks, #3394 is another example of this crash on Mojave. Apparently really not Catalina specific.

lukasmatena, 2020-01-06:
It should be fixed by @YuSanka with 4f1f507. You can retest that commit or wait for the next release.

Closing the issue, thanks for reporting.

closed 2020-01-06 by lukasmatena

fitzroy99, 2020-01-08:
Program crashes every time I push 'slice now'.

lukasmatena, 2020-01-09:
@fitzroy99 This shall be discussed in the separate issue that you opened already (https://github.com/prusa3d/PrusaSlicer/issues/3496)