prusa3d/PrusaSlicer #3373: Model slices incorrectly. Should be hollow, slices as partly solid.

duncanford, 2019-12-23:
### Version
2.1.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
E3D Toolchanger with RRF 2.04 (2019-11-01b1)

### Behavior
The model, generated with OpenSCAD, shows correctly in the plater, but is sliced as partially solid.
[image]

* _Steps needed to reproduce the problem_
Press slice

* _Expected Results_
Captured from S3D
[image]

* _Actual Results_
Captured from PrusaSlicer
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[ScrewConverter.zip](https://github.com/prusa3d/PrusaSlicer/files/3994407/ScrewConverter.zip)

neophyl, 2019-12-23:
Just tried your 3mf file in 2.2.0 alpha2 on a windows 10 machine, it reports Auto-repaired 8 errors in  the Molton Brown stl and slices correctly.  It doesnt have the partially filled interior layer anyway.
The errors were 4 degenerate facets and 4 facets removed

2.1.1 and 2.1.0 also slices correctly on my machine.  Both versions still auto repair the 8 errors in the file too.  So I'm not too sure why its not slicing for you.

this is from 2.1.0
[image]

duncanford, 2019-12-23:
Awesome. Thanks for checking. Perhaps I need to enable auto-repair ?

duncanford, 2019-12-23:
I tried repairing the STL into an OBJ file. PrusaSlicer reports object repaired, but then on import, the object produces the same error.

Given that it works for you and not for me with the same 3MF file, seems the problem is not in the 3MF file itself, but with a Slicer-level preference not stored in the file, or perhaps some other minor difference in the version. For reference, mine says at on the titlebar:

[image]

duncanford, 2019-12-23:
I've now loaded this 3MF into a fresh install of 2.2.0-alpha2+win64 and firstly I see that it now slices correctly, and secondly I also see the error report, which wasn't previously shown. In the version of 2.1.0, the model list shows:

[image]

But in 2.2.0 the model list shows:

[image]

duncanford, 2019-12-23:
Intrigued, I exported my config bundle and imported this into 2.2.0 and then re-trod the original steps taken.

1. Add slice - 0.3.stl
2. Add MoltonBrown-BigToSmall.stl

The errors that are automatically fixed when importing the project are not fixed at the time the STL is imported. Is there a way I can enable the repair?

[Models-Issue-3373.zip](https://github.com/prusa3d/PrusaSlicer/files/3995040/Models-Issue-3373.zip)

neophyl, 2019-12-23:
As far as I'm aware the files are auto repaired if you are on a windows 10 machine.  Earlier version of windows and mac/linux have to repair through netfabb.  There isnt a setting to set.

I just tried loading the stl files from your latest zip into 2.1.0 +win64-201909160915 and yes I get the same no warning on the file and the incorrect slice too. 
From reading around some of the other issues there is a vertices not quite correct issue that lots of slicers handle but Prusa Slicer doesnt.  The question is why doesn't the slicer detect/repair it initially but does after a save then import.
Bit beyond me :)  but lets hope your issue with the attached files helps the devs sort it out.

duncanford, 2019-12-23:
Logically, the reason must be that the files is not considered to be in error when imported from STL, but is considered to be in error when imported from whatever format the 3MF saves the file to.

Separate question now (not related to Prusa) is if I can alter the OpenSCAD code to not export an STL with errors in the first place.

Thanks @neophyl for sharing your time.

duncanford, 2019-12-23:
Final update for now, as I have a workaround. Noticing that after the model is recognised as having errors from the 3MF format and reports that Netfabb can auto-fix, I imported the model to Netfabb independently (https://service.netfabb.com/service.php). Once fixed by that service and re-imported, the model slices as expected. I guess the other workaround would be to export as 3MF and re-import.

rtyr, 2020-05-19:
You can force Netfabb repair even if the model is not reported as broken.

[image]

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo