prusa3d/PrusaSlicer #3374: Freezing application at startup (Win10) [2.2.0 - alpha2]

Revilo91, 2019-12-23:
### Version
PrusaSlicer-2.2.0-alpha2+win64-201912221340

### Operating system type + version
Microsoft Windows 10 Home
10.0.17763 Build 17763

### Behavior
* _Describe the problem_
Freezes on startup, no user input possible
* _Steps needed to reproduce the problem_
Try to start the alpha version of PrusaSlicer
* _Expected Results_
Start the application and work with it
* _Actual Results_
Freezing application. Cursor is a "loading icon".
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
[image]

bubnikv, 2019-12-23:
Usually a faulty OpenGL driver is guilty. Is PrusaSlicer 2.2.1 running fine on your computer?

Revilo91, 2019-12-23:
If you mean 2.1.1: This version works fine. No Problem with OpenGL 

In another issue ([https://github.com/prusa3d/PrusaSlicer/issues/2939](url)) I read about the OpenGL problem and tried to fix it, but unfortunately found no improvement.

Sandeant, 2020-01-22:
Hi all, I have the same issue with  2.1.1 and 2.2.0-alpha2 and and with older versions too
 
Operating system type + version
Microsoft Windows 10 PRO
Build 18363

PrusaSlicer startes and freezes, it doesn't matter which version of the slicer I can try to use. Also, the version of the GPU card driver does not matter.  I use RX Radeon 5700XT and I tried Radeon R9 290x - same issue.
Once, immediately after installing the latest GPU drivers, the  2.2.0-alpha2 version started successfully (at the same time, the stable version  2.1.1  still did not start witout freezing), but the next launch, after 10 minutes, was already unsuccessful again

The screenshot was taken for 2.2.0-alpha2 version , which started one time after update GPU drivers
[image]

lukasmatena, 2023-02-08:
Do you still have this issue with [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3)?

Revilo91, 2024-01-05:
Sorry for the wait, I didn't realize there was another question here.

Everything works fine now, I haven't had a problem since then.

This issue can be closed

closed 2024-01-05 by Revilo91