prusa3d/PrusaSlicer #3375: PrusaSlicer 2.2.0~alpha2 depends on Boost1.66 (due to Beast / Base64)

pmjdebruijn, 2019-12-23:
When trying to compile PrusaSlicer 2.2.0~alpha2 on Ubuntu 18.04, it fails because it depends on base64.hpp from beast, which is only available in boost1.66 or newer.

So, please consider not using the base64 function from beast, given that this implies a fairly significant dependancy increase for something that on the surface seems so trivial.

Relevant files:

obj-x86_64-linux-gnu/src/libslic3r/CMakeFiles/libslic3r.dir/CXX.includecache:boost/beast/core/detail/base64.hpp
src/libslic3r/GCode.cpp:#include <boost/beast/core/detail/base64.hpp>
src/libslic3r/GCode.cpp:                    encoded.resize(boost::beast::detail::base64::encoded_size(png_size));
src/libslic3r/GCode.cpp:                    encoded.resize(boost::beast::detail::base64::encode((void*)&encoded[0], (const void*)png_data, png_size));
src/slic3r/GUI/GUI_App.cpp:#include <boost/beast/core/detail/base64.hpp>
src/slic3r/GUI/GUI_App.cpp:                        decoded.resize(boost::beast::detail::base64::decoded_size(row.size()));
src/slic3r/GUI/GUI_App.cpp:                        decoded.resize(boost::beast::detail::base64::decode((void*)&decoded[0], row.data(), row.size()).first);

But if you must depend on the beast base64 functions, then please do raise the dependancy in CMakeLists:

    set(MINIMUM_BOOST_VERSION "1.66.0")

xenoxaos, 2020-01-01:
I ended up just using a ppa for libboost-1.70
https://launchpad.net/~mhier/+archive/ubuntu/libboost-latest

xenoxaos, 2020-01-01:
Looks like the commit for this was https://github.com/prusa3d/PrusaSlicer/commit/de60b403471bbfc510c54dae78757131e5de3ed2

pmjdebruijn, 2020-01-01:
Yeah i backported the required libboost myself, but that's workaround, though raising a major complicated dependency like boost merely for base64 function is a bit much in my view.

Salamandar, 2020-08-20:
This is still the case, and was not fixed.

pmjdebruijn, 2020-12-18:
Most current distros now have a current enough version of boost, so that using the base64 function, should no longer be an issue for most people.

Though the CMakeLists version check should be adjusted accordingly.

See #5508

bubnikv, 2021-08-06:
Fixed with #5508 by @pmjdebruijn, thanks for your contribution.

closed 2021-08-06 by bubnikv