prusa3d/PrusaSlicer #3378: Feature Request: Change Material Settings at layer Height (color Change)

StefanX13, 2019-12-23:
This is a Feature Request for future Versions.

I use the Color Change option for a while now, and its a great and simple solution to made multi color Prints. But there is one thing that could be even better.

It would be great to have a solution to change the material Settings at a selected height. Now i had to use Materials that are near the same settings (em, tmeperatur, aso.) it would be great when it is possible to select annother material at a color change including the settings above. Its ok to change the Temperatur at the laychange manuel, but especially material with annother diameter and other em settings are hard to select on the printer display.
Thanks for reading.

Area5142, 2019-12-25:
I was looking for a similar unction to change between PETG colours with different temperature setting.

A solution is to define multiple extruders in "Printer Settings" and check "Single Extruder Multi Material" as there is only one physical extruder. Each extruder can have material of your choice and be assigned to the model.
You need to be in Expert mode to use this setting.

I wish there was an easier way to do it, like the M600 colour change with a material selection.

terribleperson, 2019-12-29:
I'm another user of Prusa Slicer that would find this feature useful. The new addition of easily accessible layer-based custom g-code insertion is fantastic, but the ability to have different material profiles at different heights either through the color change dialog or the layer-based editing dialog would be extremely useful to me.

Such a feature would be particularly useful for prints that incorporate materials with substantially different maximum volumetric rates and retraction behaviour, e.g. ABS and TPE in one print.

tigert, 2019-12-30:
What if the material library also contained the color selections for different brands of filaments?
So you could use "Prusament PLA Mystic Green" in the filament selector.

For multicolor prints this would make even more sense. Instead of the color picker you could just have a material picker. So layers 0-200 use "Generic PLA Red" and then print 201-210 with "Filamentum Flexfill 98A Signal Yellow".

You would see the preview correctly, but more importantly, it would set the print temperatures and speeds so that you get a successful multimaterial print without having to type in manual gcode?

StefanX13, 2019-12-30:
@tigert thats what i mean. Like the olor Picker in the 2.2.0 Alpha, you have a Filament picker and the various Filament Settings are used in the Print for different Layers

tigert, 2020-03-25:
Yeah. This would be very useful for multimaterial prints that have a rigid and flexible part fused together.

bcwadell, 2020-04-11:
Agree this is useful, I'd like to print TPU and then overprint that with PETG layers for stiff areas of the object.  This feature would make that easy.  (I will also look at the workaround mentioned by Area5142

rtyr, 2021-02-04:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2002. Closing.

closed 2021-02-04 by rtyr