prusa3d/PrusaSlicer #3383: Huge first layer after raft

italocjs, 2019-12-24:
### Version
2.2.0 alpha, but also happened on the 2.0.0 official release

### Operating system type + version
Windows 10, 1908 - 64bits 

### 3D printer brand / version + firmware version (if known)
not related to printer

### Behavior
When creating rafts, the first layer after the raft is aroud 0.45mm,  which is higher than my nozzle, and its causing printings to look terrible from below ([print screen here](http://prntscr.com/qeyttf))

I tried different stls, all happened the same.  my "bundle" is also attached if someone has problems reproducing it 
[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/3996911/PrusaSlicer_config_bundle.zip)

I searched for option on how to manage this, but did not found anything

dompe-9, 2020-03-20:
I have the same problem.

bubnikv, 2020-03-21:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3425

closed 2020-03-21 by bubnikv