prusa3d/PrusaSlicer #3385: 3d-connextion sensitivity too low

henrikkoch, 2019-12-24:
### Version
2.2.0 alpha 2 and 1

### Operating system type + version
Window 10 64 bit

### 3D printer brand / version + firmware version (if known)
mk3s

### Behavior
when using the nice new 3d-connexion interface to steer around the plate it's very slow compared to the normal mouse interface. 

Request:  please add som multiplication on the x,y,x so it get more responsive as the normal mouse. Maybe a setting where you can do it. 

I have already tried via the control panel in 3x-connexion SW.

bubnikv, 2019-12-24:
Try the Ctrl-M short key at the 3D scene. It is even mentioned in the change log.

henrikkoch, 2019-12-24:
I now realized you in fact have added such a feature. I now only wish it to still be more sensitive. Can it be set from 1-5 please :-)

foreachthing, 2020-01-07:
This is still waaaaaaay to slow (SpacePilot Pro)!
And why can't I use the 3DConnexion settings to adjust the speed? Better: Speed settings in the 3DConnexion panel has no influence over speed in PS... :-(

enricoturri1966, 2020-01-07:
@foreachthing
We are currently by-passing the 3DConnexion driver because we cannot access it in code, that's why nothing changes in PrusaSlicer when you create a profile for it using the driver software. We tried to contact 3DConnexion to get their SDK but we did not get any answer yet.

foreachthing, 2020-01-07:
@enricoturri1966 Ok! Thanks for the info! 
I don't have an account here: https://www.3dconnexion.de/service/software-developer.html. Isn't there any help at all?

bubnikv, 2020-01-09:
They are a weird company. AFAIK they are owned by Logitech, which is Swiss,
but they do not respond at all to the software developer registrations. I
have a feeling they don't care about the current and prospective customers.
I called the support today, let's keep the fingers crossed.

foreachthing, 2020-01-09:
Yeah, well! So be it. It works as it is and my reach to the SpacePilot doesn't end in frustration (that it doesn't work) :-)
I am really grateful for interlarding 3Dx-Support!!

lukasmatena, 2020-01-31:
I consider this solved by 8559360 (for 2.2.0-alpha3 release). If anyone disagrees, we can reopen the issue. Closing it now.

closed 2020-01-31 by lukasmatena

dartrax, 2021-11-21:
I'd actually like to have it even more sensitive. I've already maxed out the "Translation" speed, but it is still too slow for me. Translation feels slower than Rotation, even though Rotation is only set to 3 and Translation already 10.
Could the Max please raised to 20 or 30? So that I could match it to the same speed as I'm used to in Autodesk Inventor.

Thanks.

lukasmatena, 2021-11-22:
@dartrax If you are the only one, instead of adjusting the range, I'll tell you a secret. It should be possible to enter any value you want by clicking the slider while holding Ctrl. We have made this more obvious in 2.4.0-beta2, where we added a small button next to each slider to do the same thing:
[image]

dartrax, 2021-11-22:
Hi @lukasmatena,
that sounds great! Indeed I can enter a value, but after moving Focus or just close, it reverts back to the maximum of 10. Are you sure it allows a value that is outside of the limits of the slider?
I also tried to change it in the PrusaSlicer.ini before starting the application, but this has no effect and gets reverted to 10 after closing the application.
[image]

lukasmatena, 2021-11-22:
To be honest, I only checked briefly, maybe that the settings are clamped somewhere else too. We'll check.

foreachthing, 2021-11-22:
Ho! What happend in 2.4.0b1? :nauseated_face:
[image]

@dartrax , I don't know about your device, but with my settings it's already really fast....?! Odd, isn't it?

dartrax, 2021-11-22:
Hi @foreachthing,
I've tried in PrusaSlicer 2.3.0, and you are right, there happend something to 2.4.0b1! In 2.3.0 I can actually set a higher value and it is saved and used.
The difference between 10 and 20 was smaller than I expected, so I tried 100 to see a difference, and now, well, 100 is really really fast ;-D
After playing with some values, I prefer something between 30 and 40 for Translation speed. This matches to a Rotation speed of 3 (now it feels like translation and rotation have the same speed/sensitivity). May be there is some factor 10 in my device somewhere.

This is my device:
[image]

And this my preferred setting in 2.3.0, that can't be used anymore in 2.4.0b1:
[image]

lukasmatena, 2021-11-22:
@dartrax The value was indeed clamped, just not where I was looking. I have increased the maximum to 30 in 4c89a9e. I'm quite puzzled about how different values you guys use and what is the source of it.

@foreachthing The layout of the dialog is indeed broken, @enricoturri1966 will look into it.

dartrax, 2021-11-22:
Here is a small comparison video. I tried to translate left and right with the same force in
1. Inventor, 
2. PrusaSlicer 2.3.0 Translation Speed = 30,
3. PrusaSlicer 2.4.0b1 Translation Speed = 3.1
As you can see, with 3.1 I'm not really able to move it...

https://user-images.githubusercontent.com/15105943/142896052-a9f1d379-b42b-45f5-8cac-b998a60dfa6d.mp4

foreachthing, 2021-11-22:
Actually, you can use the speed slider from the 3Dx menu to further increase sensitivity.
Also, in <Advanced> => move the slider more to the right. But, If you're happy with the settings in PS, it doesn't matter where you set you speed, I guess :-)

dartrax, 2021-11-22:
You mean this https://github.com/prusa3d/PrusaSlicer/issues/3385#issuecomment-975586480 setting? As you can see by the first picture, that‘s already at maximum ;-) Also in the more > submenu it‘s at max too.
But I don‘t think that this changes anything. It‘s like being ignored, I thought that was the reason for prusas own dialog ctrl+m.

foreachthing, 2021-11-23:
@dartrax it sometimes actually works (on my computer at least)! I think the values are not saved, and that's really annoying. So, yes, I'm glad for the settings in PS.

_Im Menu [Erweiterte Einstellungen] hat es weitere Schieber für die Empfindlichkeit - bei mir werden diese Einstellungen jedoch auch nicht gespeichert._

dartrax, 2021-11-27:
Oh yes I was actually talking about that advanced menu too. Mixed it up with "more" somehow while writing.
I did not recognized that it doesn't get saved, the sliders always show up with the correct setting when I go into that menu. But, it has no effect, but I'll observe that, may be it works sporadically like you said, and I'll also try on our work PCs, they have the same 3D mouses and the same slow translation.

I the meantime I'm very happy with the ctrl+m Menu of Prusa, it's working great in beta 2! Thanks @lukasmatena!

ahs3n, 2026-08-30:
When pressing Ctrl + M in 2.9.6 I don't get any menu popup. Has this feature not been released yet, or is this a bug?