prusa3d/PrusaSlicer #3386: PrusaSlicer collapses when slicing large file with 0.3mm DRAFT ,Generic Flex setup

zoltan-l, 2019-12-24:
### Version
_Version of PrusaSlicer used goes here_
PrusaSlicer version 2.1.0 + win 64

### Operating system type + version
Windows 10 Home
Version 1903
Build 18362.535

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S FW 3.8.1

### Behavior
After slicing a large file with the below set up the application collapses.
Observerd and reported by user having also Win 8.1 see Forum post
https://forum.prusaprinters.org/forum/ceske-forum-obecna-diskuze-oznameni-a-novinky/prusaslicer-chyba-slicovani/#post-181042 

Verified by myself on the above configuration

### Describe the problem_

Printer setup : 0,3 mm Draft MK3
Filament : Generic Flex
Sliced  file : more than 2 MB large
After pushing the Slicing button the slicing is progressing ,at almost the end the application crashes._ 
 

#### Project File (.3MF) where problem occurs

bubnikv, 2020-01-02:
Please attach a 3MF to reproduce the issue. Thanks.

zoltan-l, 2020-01-02:
Se please the attached file.
It also collapses in several layers height it only works for 0.10 and 0.15.
My guess is , there is something wrong with Generic Flex implementation. Did not looked at the code yet.
Attached find please the project file
[Spicka.zip](https://github.com/prusa3d/PrusaSlicer/files/4016124/Spicka.zip)

lukasmatena, 2020-01-02:
As far as I can see, your 3MF has neither 0.3 DRAFT print profile not Generic FLEX selected, and it sliced fine on my Linux setup, even after I selected them. Can you confirm that the 3MF that you posted indeed crashes when you load and slice it? It can of course be Windows-only bug.

zoltan-l, 2020-01-02:
sorry, wrong file attached. Now it should be the correct one - could be Windows only issue
[Spicka_flex.zip](https://github.com/prusa3d/PrusaSlicer/files/4016306/Spicka_flex.zip)

lukasmatena, 2020-01-02:
It indeed crashes in 2.1.0, but not in 2.2.0-alpha2. Most likely some bug that was fixed already (don't know which one). Could you retest and close the issue if happy with the result?

closed 2020-01-02 by zoltan-l