prusa3d/PrusaSlicer #3387: Export G-Code to USB drive

Area5142, 2019-12-25:
### Version
Version: 2.2.0-alpha1+linux-x64, Version: 2.2.0-alpha2+linux-x64

### Operating system type + version
Linux Mint 19.3 64 bit

### Behavior
When exporting the G-Code, the new PrusaSlicer version forces selection of the USB drive if attached. This happens to be my permanent attached backup USB drive, which is never used for G-Code. I see no way to turn this behaviour off.

My normal workflow is:
- Import, setup and prepare model (STL) in PrusaSlicer.
- Export G-Code to same working directory as the model/3mf resides (for later use).
- Get USB/SD drive or use network (FlashAir) to transfer a copy of the G-Code to printer using Explorer.

Please make an option to disable this feature or select model/3mf directory for export of G-Code file.
It is cumbersome to each time retrace back to the original model/3mf directory, just to save the G-Code.

bubnikv, 2020-01-02:
We should do something about it.

lukasmatena, 2020-01-31:
This was improved by @kocikdav for the next release (2.2.0-alpha3). The export button is duplicated in case a removable drive is connected. Each button defaults to its last used location.

I'm closing the issue. Test the new version after it is released and in case you don't like something about the way it is implemented, we can reopen it.

closed 2020-01-31 by lukasmatena

Area5142, 2020-02-01:
I am eager to try out this solution, but version 2.2.0 alpha 3 is provided only as source, not as compiled version (last 7 days).

lukasmatena, 2020-02-01:
That's why I wrote "after it is released". It might be next week or the one after that.