prusa3d/PrusaSlicer #3388: 3DConnexion SpaceNavigator USB not working on Linux Mint 19.3

Area5142, 2019-12-25:
### Version
Version: 2.2.0-alpha1+linux-x64, Version: 2.2.0-alpha2+linux-x64

### Operating system type + version
Linux Mint 19.3 64 bit

3DConnexion device:
Product: SpaceNavigator USB
New USB device found, idVendor=046d, idProduct=c626

### Behavior
File:  90-3dconnexion.rules copied to /dev/udev as instructed in release notes.
No reaction on device.

I know that the release notes states, that only some of the Linux installations works.
In this configuration the 3D mouse is not working.

bubnikv, 2020-01-02:
You have to have libudev installed.

Do you get any error reports on the console?

Area5142, 2020-01-03:
This is the console output when executing the alpha-2 version:
[console_output.txt](https://github.com/prusa3d/PrusaSlicer/files/4017768/console_output.txt)

I installed the 3DConnexion Linux x64 drivers from the web site. Do I have to install more?

bubnikv, 2020-01-03:
>  I installed the 3DConnexion Linux x64 drivers from the web site.

From their web site?

AFAIK there are two ways of accessing the HID (Human Interface Device)
devices from Linux: Either accessing low level USB (using libusub), or by
accessing a HID driver (using HIDRAW interface
https://www.kernel.org/doc/html/latest/hid/hidraw.html). I am not sure what
is happening on your system, but PrusaSlicer tries to access the HIDRAW
interface. It may very well be, that the 3DConnexion driver accesses the
low level USB interface, thus the HIDRAW does not work for PrusaSlicer, but
that is just a wild guess.

We will have to do some more experiments on Linux with the 3DConnexion
device. I am not developing on Linux regularly, but our two other colleges
do, so it will be easier for them to troubleshoot this issue.

Area5142, 2020-01-03:
I use both Windows and Linux and mixed them up, sorry. Windows drivers are from 3DConnexion web site, BUT the Linux drivers is the free software from SourceForge: [http://spacenav.sourceforge.net/](http://spacenav.sourceforge.net/)
Can be installed from Ubuntu repository: `apt install spacenavd`

The drivers/SDK on SourceForge is a free alternative to the official SDK from 3DConnexion and supports GNU/Linux, FreeBSD and MacOS X. It is difficult to get access to the official SDK and that is properly why this open source alternative exists in the first place.

Blender supports 3DConnexion devices, is open source and works on Windows, Linux and MacOS - maybe you could borrow some code snippets from their development to get a stable code base.
As I read the Blender documentation it communicates directly with the devices.

Area5142, 2020-02-18:
OS: Linux Mint 19.3
Version: PrusaSlicer-2.2.0-alpha4+linux-x64-202002101524.AppImage
3DConnexion Space Mouse connected via USB.

Still no 3D Connexion Space Mouse Detection. Starting with --loglevel=3 give following output:

```
$ ./PrusaSlicer-2.2.0-alpha4+linux-x64-202002101524.AppImage --loglevel=3
[2020-02-18 02:47:43.391756] [0x00007f0598f52540] [info]    Checking if indices need to be installed from resources...
[2020-02-18 02:47:43.687691] [0x00007f0598f52540] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 61MB; Shared memory: 44MB; Private memory: 17MB; Peak memory usage: 61MB
[2020-02-18 02:47:45.556751] [0x00007f0598f52540] [info]    Checking for cached configuration updates...
[2020-02-18 02:47:45.556776] [0x00007f0598f52540] [info]    Confing bundle not installed for vendor %1%, skipping: Creality
[2020-02-18 02:47:45.559375] [0x00007f0598f52540] [info]    No configuration updates available.
[2020-02-18 02:47:45.584674] [0x00007f0578a53700] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
[2020-02-18 02:47:45.584725] [0x00007f0578a53700] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`: HTTP 0, Unsupported protocol:
Protocol "https" not supported or disabled in libcurl
[Error 1]
[2020-02-18 02:47:45.584737] [0x00007f0578a53700] [info]    Syncing configuration cache
[2020-02-18 02:47:45.584740] [0x00007f0578a53700] [warning] No such vendor: Creality
[2020-02-18 02:47:45.584743] [0x00007f0578a53700] [info]    Downloading index for vendor: Prusa Research
[2020-02-18 02:47:45.584750] [0x00007f0578a53700] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`
	-> `/home/zzz/.PrusaSlicer-alpha/cache/PrusaResearch.idx-update`
	via tmp path `/home/zzz/.PrusaSlicer-alpha/cache/PrusaResearch.idx-update.17916.download`
[2020-02-18 02:47:45.707957] [0x00007f0578a53700] [error]   Error getting: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`: HTTP 0, Unsupported protocol:
Protocol "https" not supported or disabled in libcurl
[Error 1]
[2020-02-18 02:47:46.573861] [0x00007f0598f52540] [error]   Unable to enumerate HID devices
[2020-02-18 02:47:48.661047] [0x00007f0598f52540] [error]   Unable to enumerate HID devices
[2020-02-18 02:47:50.692735] [0x00007f0598f52540] [error]   Unable to enumerate HID devices
[2020-02-18 02:47:52.701391] [0x00007f0598f52540] [error]   Unable to enumerate HID devices
[2020-02-18 02:47:54.700907] [0x00007f0598f52540] [error]   Unable to enumerate HID devices
... continues
```
No mention of the 3D Space Navigator mouse or failure to communicate with it.

bubnikv, 2020-02-18:
will it work if you start slicer as a root?

Area5142, 2020-02-18:
Running the application as root gives the same result.

bubnikv, 2020-02-18:
What is your exact model of the 3D mouse? We have some 4 different models
and one of them keeps disconnecting on Linux, while it works on Windows and
OSX.

Area5142, 2020-02-18:
The device has no specific model information on the physical device.
3DConnexion link: https://www.3dconnexion.eu/spacemouse_compact/eu/

From Linux Mint I have collected output from _lsusb, dmesg and usb-devices_ and USBDView (Windows png) attached in the archive: _SpaceNavigator Info.gif_
[SpaceNavigator Info.zip](https://github.com/prusa3d/PrusaSlicer/files/4220193/SpaceNavigator.Info.zip)
.
Let me know if other information is needed.

bubnikv, 2020-02-18:
@kocikdav Which 3Dconnexion device keeps disconnecting on Linux?

bubnikv, 2020-02-18:
BTW is any of your linux applications working with the 3D mouse on your
computer?

Area5142, 2020-02-18:
My Space Navigator works fine with Blender version 2.8x.

bubnikv, 2020-02-19:
PrusaSlicer tries to load one of the following libraries:
"libudev.so.1", "libudev.so.0"

do you have any of the two installed in your system?

Area5142, 2020-02-19:
Yes, I found this file installed on the system:
`/lib/x86_64-linux-gnu/libudev.so.1`

bubnikv, 2020-02-20:
Could you please verify that this libudev.so.1 is loaded by PrusaSlicer
from there?

https://jvns.ca/blog/2014/03/10/debugging-shared-library-problems-with-strace/

Thanks

Area5142, 2020-02-20:
I'm not sure how to do that, but running strace on PrusaSlicer I get a very large logfile/dump. In this file a section matched libudev:
```
strace -o strace.txt ./PrusaSlicer-2.2.0-alpha4+linux-x64-202002101524.AppImage
```
```
openat(AT_FDCWD, "tls/haswell/x86_64/libudev.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "tls/haswell/libudev.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "tls/x86_64/libudev.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "tls/libudev.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "haswell/x86_64/libudev.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "haswell/libudev.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "x86_64/libudev.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "libudev.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/tmp/.mount_PrusaSNkVD3W/usr/bin/bin/libudev.so.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 14
fstat(14, {st_mode=S_IFREG|0644, st_size=125730, ...}) = 0
mmap(NULL, 125730, PROT_READ, MAP_PRIVATE, 14, 0) = 0x7efe135b2000
close(14)                               = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libudev.so.1", O_RDONLY|O_CLOEXEC) = 14
read(14, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\3008\0\0\0\0\0\0"..., 832) = 832
fstat(14, {st_mode=S_IFREG|0644, st_size=121016, ...}) = 0
mmap(NULL, 2218248, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 14, 0) = 0x7efe0593a000
mprotect(0x7efe05957000, 2093056, PROT_NONE) = 0
mmap(0x7efe05b56000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 14, 0x1c000) = 0x7efe05b56000
close(14)                               = 0
```
One line looks like a success full open of library libudev.so.1.
Please let me know if I have to run the _strace_ command with specific parameters and/or what part of the output you like to see.

Some errors is written to the console when PrudsSlicer is closed:
```
[2020-02-20 14:01:04.127894] [0x00007f6d8434a540] [error]   Unable to enumerate HID devices
[2020-02-20 14:01:04.149016] [0x00007f6d6009a700] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`: HTTP 0, Unsupported protocol:
Protocol "https" not supported or disabled in libcurl
[Error 1]
[2020-02-20 14:01:04.213193] [0x00007f6d6009a700] [error]   Error getting: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`: HTTP 0, Unsupported protocol:
Protocol "https" not supported or disabled in libcurl
[Error 1]

(prusa-slicer:5096): GLib-GObject-WARNING **: 14:01:06.848: g_object_weak_unref: couldn't find weak ref 0x7f6d80534e00(0x42e9ac8)

(prusa-slicer:5096): GLib-GObject-WARNING **: 14:01:06.867: g_object_weak_unref: couldn't find weak ref 0x7f6d80534e00(0x365e6c8)

(prusa-slicer:5096): GLib-GObject-WARNING **: 14:01:06.873: g_object_weak_unref: couldn't find weak ref 0x7f6d80534e00(0x38f3b38)

(prusa-slicer:5096): GLib-GObject-WARNING **: 14:01:06.895: g_object_weak_unref: couldn't find weak ref 0x7f6d80534e00(0x2bbb1c8)

** (prusa-slicer:5096): CRITICAL **: 14:01:06.906: gail_notebook_real_remove_gtk: assertion 'obj' failed

** (prusa-slicer:5096): CRITICAL **: 14:01:06.908: gail_notebook_real_remove_gtk: assertion 'obj' failed
```
Not sure if they are important.

bubnikv, 2020-02-20:
thanks for the report. It seems like the correct library is being loaded. I
am out of my wits.

Area5142, 2020-03-07:
OS: Linux Mint 19.3 x64
PS: PrusaSlicer-2.2.0-rc2+linux-x64-202003071144.AppImage

Just tested the new RC2 version with loglevel=3 and the result is the same:

```
$ ./prusaslicer --loglevel=3
[2020-03-07 16:42:20.576581] [0x00007f739bf65540] [info]    Checking if indices need to be installed from resources...
[2020-03-07 16:42:20.877085] [0x00007f739bf65540] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 61MB; Shared memory: 44MB; Private memory: 17MB; Peak memory usage: 61MB
[2020-03-07 16:42:21.807917] [0x00007f739bf65540] [info]    Checking for cached configuration updates...
[2020-03-07 16:42:21.807949] [0x00007f739bf65540] [info]    Confing bundle not installed for vendor %1%, skipping: Creality
[2020-03-07 16:42:21.810595] [0x00007f739bf65540] [info]    No configuration updates available.
[2020-03-07 16:42:22.603907] [0x00007f736c00e700] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
[2020-03-07 16:42:22.748415] [0x00007f736c00e700] [info]    Got PrusaSlicer online version: `2.1.1`. Sending to GUI thread...
[2020-03-07 16:42:22.749533] [0x00007f736c00e700] [info]    Syncing configuration cache
[2020-03-07 16:42:22.749552] [0x00007f736c00e700] [warning] No such vendor: Creality
[2020-03-07 16:42:22.749565] [0x00007f736c00e700] [info]    Downloading index for vendor: Prusa Research
[2020-03-07 16:42:22.749593] [0x00007f736c00e700] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`
	-> `/home/<user>/.PrusaSlicer/cache/PrusaResearch.idx-update`
	via tmp path `/home/<user>/.PrusaSlicer/cache/PrusaResearch.idx-update.10361.download`
[2020-03-07 16:42:22.880466] [0x00007f738e348700] [error]   Unable to enumerate HID devices
...
[2020-03-07 16:42:54.928109] [0x00007f738e348700] [error]   Unable to enumerate HID devices
[2020-03-07 16:42:56.931233] [0x00007f738e348700] [error]   Unable to enumerate HID devices

(prusa-slicer:10361): GLib-GObject-WARNING **: 16:42:58.037: g_object_weak_unref: couldn't find weak ref 0x7f7397f32e00(0x4ac50c8)

(prusa-slicer:10361): GLib-GObject-WARNING **: 16:42:58.055: g_object_weak_unref: couldn't find weak ref 0x7f7397f32e00(0x3e19658)

(prusa-slicer:10361): GLib-GObject-WARNING **: 16:42:58.062: g_object_weak_unref: couldn't find weak ref 0x7f7397f32e00(0x40b42c8)

(prusa-slicer:10361): GLib-GObject-WARNING **: 16:42:58.084: g_object_weak_unref: couldn't find weak ref 0x7f7397f32e00(0x33b2cc8)

** (prusa-slicer:10361): CRITICAL **: 16:42:58.096: gail_notebook_real_remove_gtk: assertion 'obj' failed

** (prusa-slicer:10361): CRITICAL **: 16:42:58.098: gail_notebook_real_remove_gtk: assertion 'obj' failed

```
Could it be the error: Unable to enumerate HID devices
That is the reason for PrusaSlicer not detecting the 3DConnexion 3DNavigator?

Area5142, 2020-03-11:
OS: Linux Mint 19.3 x64
PS: PrusaSlicer-2.2.0-rc3+linux-x64-202003111835.AppImage

Just tested the new RC3 version with loglevel=3 and the result is the same:
```
$ ./PrusaSlicer-2.2.0-rc3+linux-x64-202003111835.AppImage --loglevel=3
[2020-03-12 00:53:44.559281] [0x00007f85068f0540] [info]    Checking if indices need to be installed from resources...
[2020-03-12 00:53:44.857867] [0x00007f85068f0540] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 61MB; Shared memory: 44MB; Private memory: 17MB; Peak memory usage: 61MB
[2020-03-12 00:53:46.568114] [0x00007f85068f0540] [info]    Checking for cached configuration updates...
[2020-03-12 00:53:46.568146] [0x00007f85068f0540] [info]    Confing bundle not installed for vendor %1%, skipping: Creality
[2020-03-12 00:53:46.568164] [0x00007f85068f0540] [info]    Confing bundle not installed for vendor %1%, skipping: BIBO2Touch
[2020-03-12 00:53:46.570867] [0x00007f85068f0540] [info]    No configuration updates available.
[2020-03-12 00:53:46.579161] [0x00007f84d714e700] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
[2020-03-12 00:53:46.737737] [0x00007f84d714e700] [info]    Got PrusaSlicer online version: `2.1.1`. Sending to GUI thread...
[2020-03-12 00:53:46.739934] [0x00007f84d714e700] [info]    Syncing configuration cache
[2020-03-12 00:53:46.739970] [0x00007f84d714e700] [warning] No such vendor: Creality
[2020-03-12 00:53:46.739985] [0x00007f84d714e700] [warning] No such vendor: BIBO2Touch
[2020-03-12 00:53:46.739998] [0x00007f84d714e700] [info]    Downloading index for vendor: Prusa Research
[2020-03-12 00:53:46.740065] [0x00007f84d714e700] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`
	-> `/home/<user>/.PrusaSlicer/cache/PrusaResearch.idx-update`
	via tmp path `/home/<user>/.PrusaSlicer/cache/PrusaResearch.idx-update.14257.download`
[2020-03-12 00:53:46.860582] [0x00007f84f8cd3700] [error]   Unable to enumerate HID devices

(prusa-slicer:14257): GLib-GObject-WARNING **: 00:53:48.899: g_object_weak_unref: couldn't find weak ref 0x7f85028bde00(0x5363758)

(prusa-slicer:14257): GLib-GObject-WARNING **: 00:53:48.916: g_object_weak_unref: couldn't find weak ref 0x7f85028bde00(0x46c0cc8)

(prusa-slicer:14257): GLib-GObject-WARNING **: 00:53:48.922: g_object_weak_unref: couldn't find weak ref 0x7f85028bde00(0x49550c8)

(prusa-slicer:14257): GLib-GObject-WARNING **: 00:53:48.944: g_object_weak_unref: couldn't find weak ref 0x7f85028bde00(0x3c53ec8)

** (prusa-slicer:14257): CRITICAL **: 00:53:48.955: gail_notebook_real_remove_gtk: assertion 'obj' failed

** (prusa-slicer:14257): CRITICAL **: 00:53:48.957: gail_notebook_real_remove_gtk: assertion 'obj' failed

```

bubnikv, 2020-03-12:
Please test this build, it does not fix your issue, but it should print a
little bit more info to the console.
https://www.dropbox.com/s/btk1blf1gpi37et/PrusaSlicer-2.2.0-rc3%2B15-linux-x64-g6866c2b-202003121402.AppImage?dl=0

Area5142, 2020-03-12:
This is the output from PS test version: PrusaSlicer-2.2.0-rc3+15-linux-x64-g6866c2b-202003121402.AppImage
```
$ ./PrusaSlicer-2.2.0-rc3+15-linux-x64-g6866c2b-202003121402.AppImage --loglevel=3
[2020-03-12 16:38:21.108571] [0x00007f03092f5540] [info]    Checking if indices need to be installed from resources...
[2020-03-12 16:38:21.108741] [0x00007f03092f5540] [info]    Install index from resources: "BIBO.idx"
[2020-03-12 16:38:21.109548] [0x00007f03092f5540] [info]    Install index from resources: "LulzBot.idx"
[2020-03-12 16:38:21.412123] [0x00007f03092f5540] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 61MB; Shared memory: 44MB; Private memory: 17MB; Peak memory usage: 61MB
[2020-03-12 16:38:23.158954] [0x00007f03092f5540] [info]    Checking for cached configuration updates...
[2020-03-12 16:38:23.159083] [0x00007f03092f5540] [info]    Confing bundle not installed for vendor Creality, skipping: 
[2020-03-12 16:38:23.159150] [0x00007f03092f5540] [info]    Confing bundle not installed for vendor BIBO, skipping: 
[2020-03-12 16:38:23.159205] [0x00007f03092f5540] [info]    Confing bundle not installed for vendor BIBO2Touch, skipping: 
[2020-03-12 16:38:23.162129] [0x00007f03092f5540] [info]    Confing bundle not installed for vendor LulzBot, skipping: 
[2020-03-12 16:38:23.162135] [0x00007f03092f5540] [info]    No configuration updates available.
[2020-03-12 16:38:23.170397] [0x00007f02e9ffb700] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
[2020-03-12 16:38:23.417256] [0x00007f02fb6d8700] [error]   Unable to enumerate HID devices
[2020-03-12 16:38:23.419818] [0x00007f02e9ffb700] [info]    Got PrusaSlicer online version: `2.1.1`. Sending to GUI thread...
[2020-03-12 16:38:23.421350] [0x00007f02e9ffb700] [info]    Syncing configuration cache
[2020-03-12 16:38:23.421356] [0x00007f02e9ffb700] [warning] No such vendor: Creality
[2020-03-12 16:38:23.421360] [0x00007f02e9ffb700] [warning] No such vendor: BIBO
[2020-03-12 16:38:23.421363] [0x00007f02e9ffb700] [warning] No such vendor: BIBO2Touch
[2020-03-12 16:38:23.421367] [0x00007f02e9ffb700] [info]    Downloading index for vendor: Prusa Research
[2020-03-12 16:38:23.421382] [0x00007f02e9ffb700] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`
	-> `/home/<user>/.PrusaSlicer/cache/PrusaResearch.idx-update`
	via tmp path `/home/<user>/.PrusaSlicer/cache/PrusaResearch.idx-update.5034.download`
[2020-03-12 16:38:23.676699] [0x00007f02e9ffb700] [warning] No such vendor: LulzBot
[2020-03-12 16:38:25.420401] [0x00007f02fb6d8700] [error]   Unable to enumerate HID devices
[2020-03-12 16:38:27.421301] [0x00007f02fb6d8700] [error]   Unable to enumerate HID devices
[2020-03-12 16:38:29.424311] [0x00007f02fb6d8700] [error]   Unable to enumerate HID devices
[2020-03-12 16:38:31.427408] [0x00007f02fb6d8700] [error]   Unable to enumerate HID devices
[2020-03-12 16:38:33.430564] [0x00007f02fb6d8700] [error]   Unable to enumerate HID devices
[2020-03-12 16:38:35.433751] [0x00007f02fb6d8700] [error]   Unable to enumerate HID devices
[2020-03-12 16:38:37.436782] [0x00007f02fb6d8700] [error]   Unable to enumerate HID devices
[2020-03-12 16:38:39.439774] [0x00007f02fb6d8700] [error]   Unable to enumerate HID devices

(prusa-slicer:5034): GLib-GObject-WARNING **: 16:38:39.556: g_object_weak_unref: couldn't find weak ref 0x7f03052c2e00(0x4141c58)

(prusa-slicer:5034): GLib-GObject-WARNING **: 16:38:39.573: g_object_weak_unref: couldn't find weak ref 0x7f03052c2e00(0x34a13d8)

(prusa-slicer:5034): GLib-GObject-WARNING **: 16:38:39.579: g_object_weak_unref: couldn't find weak ref 0x7f03052c2e00(0x37367d8)

(prusa-slicer:5034): GLib-GObject-WARNING **: 16:38:39.600: g_object_weak_unref: couldn't find weak ref 0x7f03052c2e00(0x2a38ec8)

** (prusa-slicer:5034): CRITICAL **: 16:38:39.611: gail_notebook_real_remove_gtk: assertion 'obj' failed

** (prusa-slicer:5034): CRITICAL **: 16:38:39.614: gail_notebook_real_remove_gtk: assertion 'obj' failed
```

bubnikv, 2020-03-12:
It looks like the HID detection library is loaded, but for some reason no
HID device is detected.
You say the 3DConnexion works in Blender for you. I wonder what is the
output of the lsusb tool on your computer. Is the 3DConnexion listed out
and how?

bubnikv, 2020-03-12:
I am afraid I am giving up. We need a volunteer who will be able to debug this issue inside the HIDAPI library to find out what is going on. 

You may try to compile PrusaSlicer yourself and try to debug it.

The "Unable to enumerate HID devices" message is confusing at least, because the HIDAPI returns null pointer even if there is no HID device to be enumerated. We shall make it a trace level output.

Area5142, 2020-03-12:
Here the result of lsusb:
```
$ lsusb
Bus 002 Device 003: ID 1058:10b8 Western Digital Technologies, Inc. Elements Portable (WDBU6Y, WDBUZG)
Bus 002 Device 002: ID 174c:3074 ASMedia Technology Inc. ASM1074 SuperSpeed hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 012: ID 0ecd:a100 Lite-On IT Corp. LDW-411SX DVD/CD Rewritable Drive
Bus 001 Device 009: ID 0409:0059 NEC Corp. HighSpeed Hub
Bus 001 Device 013: ID 056d:0002 EIZO Corp. HID Monitor Controls
Bus 001 Device 010: ID 056d:0000 EIZO Corp. Hub
Bus 001 Device 007: ID 0409:0059 NEC Corp. HighSpeed Hub
Bus 001 Device 008: ID 2001:f103 D-Link Corp. DUB-H7 7-port USB 2.0 hub
Bus 001 Device 006: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 004: ID 174c:2074 ASMedia Technology Inc. ASM1074 High-Speed hub
Bus 001 Device 005: ID 049f:0051 Compaq Computer Corp. KU-0133 Easy Access Interner Keyboard
Bus 001 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 016: ID 8087:0aa7 Intel Corp. 
Bus 001 Device 014: ID 04a9:10ca Canon, Inc. 
Bus 001 Device 017: ID 046d:c626 Logitech, Inc. 3Dconnexion Space Navigator 3D Mouse
Bus 001 Device 015: ID 05e3:0606 Genesys Logic, Inc. USB 2.0 Hub / D-Link DUB-H4 USB 2.0 Hub
Bus 001 Device 011: ID 05e3:0606 Genesys Logic, Inc. USB 2.0 Hub / D-Link DUB-H4 USB 2.0 Hub
Bus 001 Device 002: ID 046d:c01a Logitech, Inc. M-BQ85 Optical Wheel Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
The 3DConnexion device is identified as: _Bus 001 Device 017: ID 046d:c626 Logitech, Inc. 3Dconnexion Space Navigator 3D Mouse_

I'm not sure I could debug on Linux as I have just moved from Windows to Linux under a year ago (retired 30+ years Windows developer). Where in the PrusaSlicer source is the search for 3D mouse device?

bubnikv, 2020-03-12:
You will need to compile
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

You will likely need to debug the following function:
struct hid_device_info  HID_API_EXPORT *hid_enumerate(unsigned short vendor_id, unsigned short product_id)

found in 
src/hidapi/linux/hid.c

Area5142, 2020-03-12:
This is a little too complicated for me...

Area5142, 2020-03-17:
OS: Linux Mint 19.3 x64
PS: PrusaSlicer-2.2.0-rc4+linux-x64-202003151142.AppImage
Problem with 3DConnexion device is the same.

Managed to get latest PrusaSlicer source from git and compiled it on a virgin vmWare Linux Mint 19.3 x64 image with latest updates.
I copied the file "90-3dconnexion.rules" to system location: _/lib/udev/rules.d/_

Activating "ENABLE_3DCONNEXION_DEVICES_DEBUG_OUTPUT" I can see, that the device is found:
```
Detected devices:
"VMware/VMware Virtual USB Mouse" code: 3599/3 (e0f/3) serial number: '' usage page: 0 usage: 0 interface number: 0
"3Dconnexion/SpaceNavigator" code: 1133/50726 (46d/c626) serial number: '' usage page: 12080 usage: 29557 interface number: 0

Detected 3D connexion devices:
"3Dconnexion/SpaceNavigator" code: 1133/50726 (46d/c626) serial number: '' usage page: 12080 usage: 29557
```
In "Mouse3DController.cpp" the method "bool Mouse3DController::connect_device()" is called at regular intervals and finds the USB 3D mouse, but fails to make a connection.

Playing around with the source and inserting a lot of _std::cout << "...some debugging text... " << std::endl;_, the 3D Mouse was suddenly found and recognized... and it worked...:

```
Detected devices:
"VMware/VMware Virtual USB Mouse" code: 3599/3 (e0f/3) serial number: '' usage page: 0 usage: 0 interface number: 0
"3Dconnexion/SpaceNavigator" code: 1133/50726 (46d/c626) serial number: '' usage page: 12080 usage: 29557 interface number: 0

Detected 3D connexion devices:
"3Dconnexion/SpaceNavigator" code: 1133/50726 (46d/c626) serial number: '' usage page: 12080 usage: 29557

Opening device: 46d/c626 using hid_open()
Opened device.
```
Not sure, what has changed and why the device suddenly was recognized... I will try rolling the changes back to see, if the mouse stops working again.

To me it looks like a threading/timing problem, but that is just a guess.

bubnikv, 2020-03-17:
interesting. congrats to compile from sources.

the timing issue may be due to whatever reasons, including hidapi, 3dconnexion device, kernel driver, you name it.

Area5142, 2020-03-18:
Reverted source and recompiled everything, but the problem is now gone on that Linux Mint virtual build image. 3DConnexion device is found every time.

If anyone else wants to try building the the binary for Linux, I included my script here:
[BuildPS.sh.zip](https://github.com/prusa3d/PrusaSlicer/files/4346710/BuildPS.sh.zip)

Only a simple script as I am new to development on Linux. To make a complete build:
1) Go to home directory.
2) Run _./BuildPS.sh all_

This will install minimal tool-set for compiling and making the binary. Fetch latest PrusaSlicer source from git, compile and link with static libraries. For other options run: _./BuildPS.sh usage_

I would like to make an AppImage (iso) portable version of the slicer binary file, but can't find any description on how to do it. This would allow me to move the program to another Linux PC where I first found the problem with the 3D device. Can I get some help for that?

Some description can be found on AppImage.org, but some help would be appreciated ;-)

bubnikv, 2020-03-18:
The AppImage is built on our build servers and currently the build is
pretty convoluted as we build on one docker image with centos supporting
latest devtoolkits with C++17 compilers while producing binaries compatible
with an old glibc, while the appimage is built on another docker image with
debian as the AppImage script refuses to run on the old centos. I suppose
the AppImage scripts run on Debian based distros only. Sorry, but it would
take somebody from us at least a day to extract and polish the AppImage
scripts from the build server. We will do it one day.

Area5142, 2020-03-18:
Ok, I understand.

When compiling the source on my Linux image the 3D device works as expected.
Moving the precompiled AppImage (PrusaSlicer-2.2.0-rc4+linux-x64-202003151142.AppImage) to the same Linux image, the 3D device is not working.
Somehow there is a difference between the application build on the build server and run inside the AppImage package and the binary I compile.

Is there a simple way of replacing the binary in the AppImage with the one I have compiled?

I will be reading up on the AppImage documentation...

Area5142, 2020-03-19:
I am closer to pinpoint the problem.

Managed to create an AppImage with a freshly compiled PrusaSlicer binary that is working. I now has the basis for comparing the binaries and when the 3D mouse stops working.

At this point I can conclude that the two binaries works the same (Prusa vs. mine) and rule out differences in the build environment. The problem is in the AppImage:

- Running slicer as AppImage => Failure to detect 3D mouse.
- Running slicer via AppRun in unpacked AppImage => Failure to detect 3D mouse.
- Running slicer as standalone program => Success to detect 3D mouse.

This is true for both versions of the binaries.

I will now focus on why the AppImage is blocking 3D mouse detection.
Without documentation of the files in the AppImage, this will be try and error...

lukasmatena, 2020-03-19:
@Area5142 I'm no expert on this, but issues like that can be caused by AppImage environment polluted with env variables that enable or disable various things. This could well be the case.

There is already at least workaround of this in PS codebase (in GUI.cpp: desktop_open_datadir_folder).

Area5142, 2020-03-19:
@lukasmatena 
I am sure the problem is in the AppImage executing environment - my tests so far show so.

The method _"void desktop_open_datadir_folder()"_ is only called from the menu and I see no other access to the environment var _"APPIMAGE"_ in the code.

bubnikv, 2020-03-19:
So if you download the linux tar.gz from our releases page and unpack, it
runs fine I suppose.

All the AppImages, FlatPacks and Snap packages are an act of violence
against the Linux systems. I find it a miracle that it somehow works.

We had to replace the standard AppRun script of AppImage with the following
to support calling perl or python G-code post processing scripts from the
host. Maybe you will find something fishy there.

#!/bin/sh
HERE="$(dirname "$(readlink -f "${0}")")"
export UNION_PRELOAD="${HERE}"
export LD_PRELOAD="${HERE}/libunionpreload.so"
export PATH="${HERE}"/usr/bin/:"${HERE}"/usr/sbin/:"${HERE}"/usr/games/:"
${HERE}"/bin/:"${HERE}"/sbin/:"${PATH}"
export LD_LIBRARY_PATH="${HERE}"/usr/lib/:"${HERE}"/usr/lib/i386-linux-gnu/:
"${HERE}"/usr/lib/x86_64-linux-gnu/:"${HERE}"/usr/lib32/:"${HERE}"
/usr/lib64/:"${HERE}"/lib/:"${HERE}"/lib/i386-linux-gnu/:"${HERE}"
/lib/x86_64-linux-gnu/:"${HERE}"/lib32/:"${HERE}"/lib64/:"${LD_LIBRARY_PATH}
"
EXEC=$(grep -e '^Exec=.*' "${HERE}"/*.desktop | head -n 1 | cut -d "=" -f
2- | sed -e 's|%.||g')
exec ${EXEC} "$@"

bubnikv, 2020-03-19:
If something went wrong, I would expect the"libudev.so.1", "libudev.so.0"
libraries to not load correctly, but this does not seem to be the case.
This libudev library may load some dependencies and configs, it will open
linux devices and possible communicate with the kernel using the online
generated files and filesystem hierarchies (I don't remember how these are
called correctly). Frankly I am out of my wits.

Area5142, 2020-03-19:
I found a quick fix, editing the AppImage startup script AppRun.

1) Unpack the AppImage:
`./PrusaSlicer-2.2.0-rc4+linux-x64-202003151142.AppImage --appimage-extract`

2) Edit the script file _squashfs-root/AppRun_ and comment out line 4 and save (_export LD_PRELOAD="${HERE}/libunionpreload.so"_). It should look like this:
```
#!/bin/sh
HERE="$(dirname "$(readlink -f "${0}")")"
export UNION_PRELOAD="${HERE}"
# export LD_PRELOAD="${HERE}/libunionpreload.so"
export PATH="${HERE}"/usr/bin/:"${HERE}"/usr/sbin/:"${HERE}"/usr/games/:"${HERE}"/bin/:"${HERE}"/sbin/:"${PATH}"
export LD_LIBRARY_PATH="${HERE}"/usr/lib/:"${HERE}"/usr/lib/i386-linux-gnu/:"${HERE}"/usr/lib/x86_64-linux-gnu/:"${HERE}"/usr/lib32/:"${HERE}"/usr/lib64/:"${HERE}"/lib/:"${HERE}"/lib/i386-linux-gnu/:"${HERE}"/lib/x86_64-linux-gnu/:"${HERE}"/lib32/:"${HERE}"/lib64/:"${LD_LIBRARY_PATH}"
EXEC=$(grep -e '^Exec=.*' "${HERE}"/*.desktop | head -n 1 | cut -d "=" -f 2- | sed -e 's|%.||g')
exec ${EXEC} "$@"
```
3) Repack the AppImage again with the tool _appimagetool-x86_64.AppImage_:
`./appimagetool-x86_64.AppImage squashfs-root`

4) The resulting AppImage will now work with the 3DConnexion device.

The _appimagetool-x86_64.AppImage_ can be downloaded from: https://github.com/AppImage/AppImageKit/releases

I have only tested this fix by moving the AppImage from the building machine to a workstation, but it fixed the 3D mouse problem.
There can be side effects not tested. I think the fix points in the right direction to find the underlying problem.

bubnikv, 2020-03-20:
We have Simon @probonopd on board, he is the original author of AppImage.
Would you Simon please comment on the fix?
Why does importing the libunionpreload.so break the hidapi library that we linking statically to communicate with the USB 3DConnexion device? For the record, our statically linked hidapi library links one of "libudev.so.1", "libudev.so.0" dynamically.

probonopd, 2020-03-20:
Not sure. libunionpreload.so is based on https://github.com/mikix/deb2snap/blob/master/src/preload.c. Predominantly it is meant as a transitional helper for applications that still have compiled-in absolute paths. If an application can be changed so that it resolves all paths relative to itself, this library should not be needed anymore. Do you think PrusaSlicer can do without it?

bubnikv, 2020-03-20:
> Do you think PrusaSlicer can do without it?

I believe so. The binary distro we had before the AppImage and which we still provide as an option relates all its data to the path of the prusa-slicer binary. Should we remove the libunionpreload from the AppRun script then?

probonopd, 2020-03-20:
Yes, give it a try and see whether it breaks anything. If not, we should be golden :-)

bubnikv, 2020-03-20:
Thanks @Area5142 and @probonopd, now we can happily close this issue.
Also PrusaSlicer will run a bit faster without the unnecessary libunionpreload.

closed 2020-03-20 by bubnikv