prusa3d/PrusaSlicer #3389: No 3DConnexion Support

facebraker, 2019-12-25:
Hello,

how can i activate the announced 3DConnexion 3D Mouse support?
I check it and nothing happened when i toggle the 3D Mouse.

Thank you and happy printing 

Alex

### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

ruedli, 2019-12-26:
For me it worked, just had the 3D connection driver installed on windows 10 and extracted the alpha 2.2.0. Could your issue be related to MacOS?

GarthSnyder, 2020-01-02:
It's not working for me, either. I have a SpaceMouse Pro Wireless, but it is actually connected with a USB cable. I'm on Catalina 10.15.2. Works fine in other apps.

ruedli, 2020-01-03:
I believe the release notes said that there were issues in this alpha
release with MacOS, being worked on...

GarthSnyder, 2020-01-03:
@ruedli, it says it should work on macOS as long as the device is hardwired, not communicating through a radio dongle. My model is one of those listed as supported.

bubnikv, 2020-01-03:
@GarthSnyder Would you please send us a console log of PrusaSlicer? Is there any mention of 
"Unable to initialize hidapi library"
"Can't create udev"
"Unable to enumerate HID devices"
?

I have asked @enricoturri1966 who is the author of the 3dconnexion integration into PrusaSlicer to add more tracing. PrusaSlicer may be started with the --loglevel={0,1,2,3,4,5} parameter, and we should log the progress of the 3Dconnexion detection for the highest loglevel.

GarthSnyder, 2020-01-03:
When I went to get console output, I realized that PrusaSlicer _is_ in fact responding to 2 out of 6 axes on the SpaceMouse: pulling up or pushing down on the puck (pans vertically) and pushing the puck to the left or right without rotation (pans horizontally). However, even on these axes the motion is very slight and has a large dead zone. (You have to push firmly to get any response.) The other axes don't seem to do anything.

Just to be clear, when you ask for a "console log of PrusaSlicer", you want All Messages from Console.app filtered by "prusa", is that right? There doesn't seem to be any dedicated log only for PrusaSlicer. That log info is attached as **console.log**. Looks OK to me except for "UNIX error exception: 17". No mention of udev or enumeration.

When I run the app with the **--loglevel=5** argument, some addtional output comes to the term window (but not the system log). It mostly does not seem relevant, except for:

[2020-01-03 00:38:05.147502] [0x0000000117f64dc0] [info]    Connected device: 3Dconnexion/SpaceMouse Pro Wireless

So at least it's seeing the SpaceMouse. Complete terminal log attached as well.

[console.log](https://github.com/prusa3d/PrusaSlicer/files/4018526/console.log)
[terminal.log](https://github.com/prusa3d/PrusaSlicer/files/4018528/terminal.log)

bubnikv, 2020-01-03:
> Just to be clear, when you ask for a "console log of PrusaSlicer", you
want All Messages from Console.app filtered by "prusa", is that right?

Cool? How do you get that log? I am currently mostly developing on Windows,
I used work a lot and I used to be a militant Linux evangelist some 20+
years ago, but use OSX sparsely.

I mant just the terminal output when running from command line, but any log
is useful.

The change log mentions a Ctrl-M hotkey to configure 3D mouse sensitivity.
We will add the zooming action to the "move forward / backward" actions.

facebraker, 2020-01-03:
@bubnikv , 
i also don't get a 3dconnextion settings dialog with the key combination Command+M.
attached my logs.

`./PrusaSlicer --loglevel=5
[2020-01-03 09:50:54.014442] [0x00000001143a75c0] [info]    Checking if indices need to be installed from resources...
[2020-01-03 09:50:54.022835] [0x00000001143a75c0] [trace]   System language detected (user locales and such): de_DE
[2020-01-03 09:50:54.023980] [0x00000001143a75c0] [trace]   Best translation language detected (may be different from user locales): de_DE
[2020-01-03 09:50:54.024014] [0x00000001143a75c0] [trace]   Switching wxLocales to de_DE
[2020-01-03 09:50:54.390988] [0x00000001143a75c0] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2020-01-03 09:50:54.391005] [0x00000001143a75c0] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2020-01-03 09:50:54.398698] [0x00000001143a75c0] [info]    Undo / Redo snapshot taken: Neues Projekt, Undo / Redo stack memory: 0MB Resident memory: 42MB; Peak memory usage: 42MB
[2020-01-03 09:50:55.921376] [0x00000001143a75c0] [debug]   TriangleMesh::repair() started
[2020-01-03 09:50:55.922056] [0x00000001143a75c0] [debug]   TriangleMesh::repair() finished
[2020-01-03 09:50:55.922069] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-01-03 09:50:55.922074] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-01-03 09:50:55.922111] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-01-03 09:50:56.042911] [0x00000001143a75c0] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2020-01-03 09:50:56.042931] [0x00000001143a75c0] [debug]   GLCanvas3D: Use Retina OpenGL: true
2020-01-03 09:50:56.045 PrusaSlicer[51603:1509821] add unmount observer
[2020-01-03 09:50:56.125248] [0x00000001143a75c0] [debug]   TriangleMesh::repair() started
[2020-01-03 09:50:56.125504] [0x00000001143a75c0] [debug]   TriangleMesh::repair() finished
[2020-01-03 09:50:56.125510] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-01-03 09:50:56.125515] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-01-03 09:50:56.125520] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-01-03 09:50:57.001535] [0x00000001143a75c0] [debug]   TriangleMesh::repair() started
[2020-01-03 09:50:57.002687] [0x00000001143a75c0] [debug]   TriangleMesh::repair() finished
[2020-01-03 09:50:57.002701] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-01-03 09:50:57.002706] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-01-03 09:50:57.002759] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-01-03 09:50:57.008146] [0x00000001143a75c0] [debug]   TriangleMesh::repair() started
[2020-01-03 09:50:57.008417] [0x00000001143a75c0] [debug]   TriangleMesh::repair() finished
[2020-01-03 09:50:57.008423] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - start
[2020-01-03 09:50:57.008426] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - stl_generate_shared_vertices
[2020-01-03 09:50:57.008434] [0x00000001143a75c0] [trace]   TriangleMeshSlicer::require_shared_vertices - end
[2020-01-03 09:50:58.540132] [0x00000001143a75c0] [info]    Checking for cached configuration updates...
[2020-01-03 09:50:58.548331] [0x00000001143a75c0] [debug]   Vendor: Prusa Research, version installed: 1.1.1-alpha2, version cached: 1.1.1-alpha2
[2020-01-03 09:50:58.548371] [0x00000001143a75c0] [info]    No configuration updates available.
[2020-01-03 09:50:58.694859] [0x00007000084d9000] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
*   Trying 217.16.180.252...
* TCP_NODELAY set
* Connected to files.prusa3d.com (217.16.180.252) port 443 (#0)
* TLS 1.2 connection using TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
* Server certificate: files.prusa3d.com
* Server certificate: Let's Encrypt Authority X3
* Server certificate: DST Root CA X3
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.2.0-alpha2+
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Fri, 03 Jan 2020 08:50:58 GMT
< Content-Length: 6
< Connection: keep-alive
< X-Accel-Version: 0.01
< Last-Modified: Fri, 03 Jan 2020 08:11:48 GMT
< ETag: "6-59b37dc7f28f7"
< Accept-Ranges: bytes
< X-Powered-By: PleskLin
<
* Connection #0 to host files.prusa3d.com left intact
[2020-01-03 09:50:58.878783] [0x00007000084d9000] [info]    Got PrusaSlicer online version: `2.1.1`. Sending to GUI thread...
[2020-01-03 09:50:58.879817] [0x00007000084d9000] [info]    Syncing configuration cache
[2020-01-03 09:50:58.879875] [0x00007000084d9000] [info]    Downloading index for vendor: Prusa Research
[2020-01-03 09:50:58.879913] [0x00007000084d9000] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`
	-> `/Users/akirchner/Library/Application Support/PrusaSlicer-alpha/cache/PrusaResearch.idx-update`
	via tmp path `/Users/akirchner/Library/Application Support/PrusaSlicer-alpha/cache/PrusaResearch.idx-update.51603.download`
[... cut ...]
< Content-Type: application/octet-stream
< Content-Length: 5824
< Last-Modified: Fri, 03 Jan 2020 08:11:48 GMT
< Connection: keep-alive
< ETag: "5e0ef744-16c0"
< X-Powered-By: PleskLin
< Accept-Ranges: bytes
<
* Connection #0 to host files.prusa3d.com left intact
[2020-01-03 09:50:58.960059] [0x00007000084d9000] [warning] The downloaded index /Users/akirchner/Library/Application Support/PrusaSlicer-alpha/cache/PrusaResearch.idx-update for vendor Prusa Research is older than the active one. Ignoring the downloaded`

GarthSnyder, 2020-01-03:
> bubnikv wrote: Cool? How do you get that log?

Just start the Console app. I think it's just a glorified interface for all of the log entries that come in through syslog (or whatever replaced syslog), regardless of the exact file they go to. There's a search bar you can use to winnow down the output.

There isn't much terminal output without the **--loglevel** argument, just a couple of lines that also appear in the console log.

I _do_ get a configuration panel with Command-M, and it lets you set the dead zone, so that issue has already been addressed. It does allow setting the sensitivity for both translation and rotation. However, rotation still doesn't doing anything even with the sensitivity turned up. In addition, the translation sensitivity is way too low even with the sensitivity turned up all the way (to 2). Are the endpoints on these sensitivity sliders set correctly?

SpaceMouses are pretty precise tools, so "normal" sensitivity is that if you push all the way left or right, the whole platter should whip right off the screen in a second or two. It's easy to use less force to get a more nuanced response.

GarthSnyder, 2020-01-03:
Forgot to mention: there is also a system-level control panel for the SpaceMouse that lets you define a profile for each app. I would imagine this is largely invisible to applications, at least as far as basic motion goes. I tried without any PrusaSlicer profile, then I created a PrusaSlicer profile and made sure all the axes were set to their default meanings. The behavior was the same both with and without a profile.

bubnikv, 2020-01-03:
> Forgot to mention: there is also a system-level control panel for the
SpaceMouse that lets you define a profile for each app. I would imagine
this is largely invisible to applications, at least as far as basic motion
goes. I tried without any PrusaSlicer profile, then I created a PrusaSlicer
profile and made sure all the axes were set to their default meanings. The
behavior was the same both with and without a profile.

I suppose the application would need to talk to the 3DConnexion driver,
which it does not as of now, as we did not have access to the drivers.

GarthSnyder, 2020-01-03:
> bubnikv wrote: I suppose the application would need to talk to the 3DConnexion driver,
which it does not as of now, as we did not have access to the drivers.

See #3447 regarding that.

lukasmatena, 2020-01-31:
@facebraker 
OSX support and extended logging will be part of upcoming 2.2.0-alpha3 release. Could you please test it after it is released and close the issue if ok? Thanks.

facebraker, 2020-02-11:
In Alpha 4 it works for me on my MacBook Pro very well.
Thank you.
Closed.

closed 2020-02-11 by facebraker