prusa3d/PrusaSlicer #3390: Feature Request - designate a part as support

jsilva-iii, 2019-12-25:
PussaSlicer-2.2.0-alpha2+win64

When you generate tree supports in external program - it get's imported as a part and as such you have to assign a filament to it.  I see in this version of slicer you can change the part type. It would be good if it could be marked as a support and so printed with whatever filament is currently loaded.

bubnikv, 2020-01-02:
If it is marked as support, shall it be printed with support style or with an object style?

jsilva-iii, 2020-01-02:
I suspect object type would be more suitable as one would want for  e.g. 2 perimeters and some infil for a tree type support  and it will print with whatever filament is currently loaded so as to avoid a filament change, I guess the exception to this would be a PVA filament as this would weaken the tree structure..

stuartpb, 2020-01-31:
> It would be good if it could be marked as a support and so printed with whatever filament is currently loaded.

Can't this be achieved by setting the "Wipe into this object" option?

wavexx, 2020-02-28:
Related:  #3117

Gregor-Gregor, 2025-03-11:
Related: [#14252](https://github.com/prusa3d/PrusaSlicer/issues/14252)

closed 2026-09-29 by prusaslicer-bot[bot]