prusa3d/PrusaSlicer #3391: Legend in sliced view shows not maximum achieved, but maximum possible.

ProfDrYoMan, 2019-12-25:
### Version
2.2.0-alpha2

### Behavior
It would be great that speeds are shown in the legend not from slowest to maximum possible, but lowest to maximum achieved after slice.

[image]

Legend shows up to 400 (which is setup by me for very small width infills on low layer height and limited only by flow).
But if the print is sliced to, let's say, maximum 80 mm/s in the fastest parts it still shows up to 400.

_Is this a new feature request?_

No.

bubnikv, 2020-01-02:
You are right. It would be nice, but much more complex to show the speeds calculated by the accurate time estimator.

ProfDrYoMan, 2020-01-02:
Well, the sliced preview already shows the colors. The legend should use the highest colored speed on top. Easy. No estimation needed.

rtyr, 2021-02-04:
Fixed. Closing.

closed 2021-02-04 by rtyr