prusa3d/PrusaSlicer #3392: Please add instructions for installing PrusaSlicer on Chromebooks with Linux Support

mrjohnc, 2019-12-25:
So I recently switched to a Chromebook with Linux Support and I know it should be possible to install and run PrusaSlicer like I've done with other Linux programs like InkScape. I do not want to install Ubuntu or other stand alone Linux distros using developer mode because I'll lose access to apps I need for work switching to developer mode.

I understand that I should be able to install it using Debian commands in the terminal or possibly by opening the tar package from GitHub but don't understand how because I'm fairly new to Linux. 

I think this would be really helpful documentation to have as a lot of people have Chromebooks.

Any help or instructions would be really appreciated.

Thanks

gzub, 2019-12-29:
I just did this as I just assembled my printer, it really is as simple as enabling linux on the chromebook, download the appimage (in files app share the download directory with linux), copying it to the user directory from the download directory, changing it to executable (chmod u+x [filename]) and running the appimage... It will only work on x86 chromebooks...

mrjohnc, 2019-12-29:
> I just did this as I just assembled my printer, it really is as simple as enabling linux on the chromebook, download the appimage (in files app share the download directory with linux), copying it [...]

Yes I also wrote some similar instructions here https://forum.prusaprinters.org/forum/prusaslicer/how-to-install-and-run-prusaslicer-on-chromebook-with-linux-support/#post-181273

However it would be very useful to have some instructions that are easy to find and not assuming people will know what Linux is, what an appimage is, how to use a commandline and what Linux commands are and how to do them.

mikolaszuza, 2020-03-17:
Hey @mrjohnc and @gzub,
starting with version 2.2, the PrusaSlicer team started publishing builds for the Raspberry Pi4.

If you're running an ARM-powered Chromebook, maybe you could use that build too and skip the Linux virtualization?
Anyway, just a wild guess.

bubnikv, 2021-04-14:
Documented, closing.
https://help.prusa3d.com/en/article/install-prusaslicer_1903

PrusaSlicer 2.3.1-rc will have some improvements in regard to ChromeOS integration, see the upcoming change log.

closed 2021-04-14 by bubnikv