prusa3d/PrusaSlicer #3393: Project (.3mf) files using custom printer profile loads with corrupted printer name.

MucaroBoricua, 2019-12-25:
### Version
PrusaSlicer version 2.1.1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Stock Original Prusa i3 MK2S running 3.2.3 firmware

### Behavior
* If a project (.3mf file) saved using a custom printer profile is loaded, the printer is set to the name of the project file followed by the name of the custom printer.  See attached pic.

#### Project File (.3MF) where problem occurs
[Sample project files.zip](https://github.com/prusa3d/PrusaSlicer/files/4000817/Sample.project.files.zip)

[image]

bubnikv, 2020-01-02:
This is the way the software indicates that the profile stored in the 3mf is different from the source profile. Is it not your case?

What do you expect the software to show?

bubnikv, 2020-03-15:
no response, closing.

closed 2020-03-15 by bubnikv