prusa3d/PrusaSlicer #3394: Crash in PrusaSlicer 2.2.0-alpha2

Daghis, 2019-12-25:
### Version
2.2.0-alpha2+

### Operating system type + version
macOS Mojave (10.14.6)

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior
1. I added an STL to the build plate. I was set to print on my MK3S MMU2S.
2. I sliced the STL.
3. I then added a color change at 2.00 mm high.
4. I went into the Printer settings and attempted to change the color for Extruder 1.
5. I changed the color to red, but when I closed the color-choosing pop-up window, PrusaSlicer crashed with the exception excerpted below (and attached in full).

[image]

```
Process:               PrusaSlicer [5749]
Path:                  /Users/USER/Desktop/PrusaSlicer-2.2.0-a2.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.2.0-alpha2+-201912221330 (PrusaSlicer-2.2.0-alpha2+-201912221330)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           PrusaSlicer [5749]
User ID:               501

Date/Time:             2019-12-25 18:44:27.859 -0500
OS Version:            Mac OS X 10.14.6 (18G1012)
Report Version:        12
Anonymous UUID:        1C3CD60A-1B09-73E9-301A-EB0B1E071784

Time Awake Since Boot: 2200000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [5749]

VM Regions Near 0:
--> 
    __TEXT                 0000000102e48000-0000000103dff000 [ 15.7M] r-x/r-x SM=COW  /Users/USER/Desktop/PrusaSlicer-2.2.0-a2.app/Contents/MacOS/PrusaSlicer

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation       0x00007fff30420457 CFDictionaryGetValue + 11
1   com.prusa3d.slic3r/            0x000000010369dcef wxMacCoreGraphicsContext::DoDrawText(wxString const&, double, double) + 319
2   com.prusa3d.slic3r/            0x0000000103724a93 wxGCDCImpl::DoDrawText(wxString const&, int, int) + 147
3   com.prusa3d.slic3r/            0x00000001032fc686 Slic3r::GUI::GLCanvas3D::LegendTexture::generate(Slic3r::GCodePreviewData const&, std::__1::vector<float, std::__1::allocator<float> > const&, Slic3r::GUI::GLCanvas3D const&, bool) + 1414
4   com.prusa3d.slic3r/            0x0000000103307b10 Slic3r::GUI::GLCanvas3D::load_gcode_preview(Slic3r::GCodePreviewData const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&) + 768
5   com.prusa3d.slic3r/            0x00000001033e1e09 Slic3r::GUI::Preview::load_print_as_fff(bool) + 1449
6   com.prusa3d.slic3r/            0x00000001033e276a Slic3r::GUI::Preview::reload_print(bool) + 74
7   com.prusa3d.slic3r/            0x00000001034227e4 Slic3r::GUI::Plater::priv::update(unsigned int) + 436
8   com.prusa3d.slic3r/            0x0000000103431c84 Slic3r::GUI::Plater::on_config_change(Slic3r::DynamicPrintConfig const&) + 3668
9   com.prusa3d.slic3r/            0x00000001034dc069 Slic3r::GUI::TabPrinter::update() + 217
10  com.prusa3d.slic3r/            0x00000001034b95d4 Slic3r::GUI::Tab::on_value_change(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, boost::any const&) + 2532
11  com.prusa3d.slic3r/            0x00000001034f8628 std::__1::__function::__func<Slic3r::GUI::Page::new_optgroup(wxString const&, int)::$_37, std::__1::allocator<Slic3r::GUI::Page::new_optgroup(wxString const&, int)::$_37>, void (std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, boost::any const&)>::operator()(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, boost::any const&) + 104
12  com.prusa3d.slic3r/            0x000000010351b616 Slic3r::GUI::ConfigOptionsGroup::on_change_OG(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, boost::any const&) + 262
13  com.prusa3d.slic3r/            0x0000000103516c9d wxEventFunctorFunctor<wxEventTypeTag<wxColourPickerEvent>, Slic3r::GUI::ColourPicker::BUILD()::$_14>::operator()(wxEvtHandler*, wxEvent&) + 93
14  com.prusa3d.slic3r/            0x00000001038b13ee wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
15  com.prusa3d.slic3r/            0x00000001038b115b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
16  com.prusa3d.slic3r/            0x00000001038b1070 wxEvtHandler::ProcessEvent(wxEvent&) + 96
17  com.prusa3d.slic3r/            0x00000001037af052 wxWindowBase::TryAfter(wxEvent&) + 146
18  com.prusa3d.slic3r/            0x00000001036ad1e0 wxGenericColourButton::OnButtonClick(wxCommandEvent&) + 416
19  com.prusa3d.slic3r/            0x00000001038b13ee wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
20  com.prusa3d.slic3r/            0x00000001038b115b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
21  com.prusa3d.slic3r/            0x00000001038b1070 wxEvtHandler::ProcessEvent(wxEvent&) + 96
22  com.prusa3d.slic3r/            0x00000001038b158c wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
23  com.prusa3d.slic3r/            0x0000000103677652 wxButton::OSXHandleClicked(double) + 162
24  com.apple.AppKit               0x00007fff2dd3b644 -[NSApplication(NSResponder) sendAction:to:from:] + 312
25  com.apple.AppKit               0x00007fff2dda5992 -[NSControl sendAction:to:] + 86
26  com.apple.AppKit               0x00007fff2dda58c4 __26-[NSCell _sendActionFrom:]_block_invoke + 136
27  com.apple.AppKit               0x00007fff2dda57c6 -[NSCell _sendActionFrom:] + 178
28  com.apple.AppKit               0x00007fff2ddd254b -[NSButtonCell _sendActionFrom:] + 96
29  com.apple.AppKit               0x00007fff2dda40e1 -[NSCell trackMouse:inRect:ofView:untilMouseUp:] + 2375
30  com.apple.AppKit               0x00007fff2ddd229c -[NSButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 698
31  com.apple.AppKit               0x00007fff2dda2b1e -[NSControl mouseDown:] + 791
32  com.prusa3d.slic3r/            0x00000001036e9075 wxWidgetCocoaImpl::mouseEvent(NSEvent*, NSView*, void*) + 261
33  com.apple.AppKit               0x00007fff2dc7e937 -[NSWindow(NSEventRouting) _handleMouseDownEvent:isDelayedEvent:] + 5724
34  com.apple.AppKit               0x00007fff2dbb51a6 -[NSWindow(NSEventRouting) _reallySendEvent:isDelayedEvent:] + 2295
35  com.apple.AppKit               0x00007fff2dbb4667 -[NSWindow(NSEventRouting) sendEvent:] + 478
36  com.prusa3d.slic3r/            0x00000001036d7adb -[wxNSWindow sendEvent:] + 123
37  com.apple.AppKit               0x00007fff2da53e4b -[NSApplication(NSEvent) sendEvent:] + 331
38  com.prusa3d.slic3r/            0x000000010365a382 -[wxNSApplication sendEvent:] + 98
39  com.apple.AppKit               0x00007fff2da425c0 -[NSApplication run] + 755
40  com.prusa3d.slic3r/            0x00000001036cdb5e wxGUIEventLoop::OSXDoRun() + 174
41  com.prusa3d.slic3r/            0x000000010389c54c wxCFEventLoop::DoRun() + 44
42  com.prusa3d.slic3r/            0x0000000103823765 wxEventLoopBase::Run() + 85
43  com.prusa3d.slic3r/            0x00000001037fe1e4 wxAppConsoleBase::MainLoop() + 116
44  com.prusa3d.slic3r/            0x0000000103691bfa wxApp::OnRun() + 26
45  com.prusa3d.slic3r/            0x000000010384eef3 wxEntry(int&, wchar_t**) + 131
46  com.prusa3d.slic3r/            0x0000000102e53219 Slic3r::CLI::run(int, char**) + 15529
47  com.prusa3d.slic3r/            0x0000000102e564e0 main + 352
48  libdyld.dylib                  0x00007fff5c3da3d5 start + 1
```

[PrusaSlicer-2.2.0-alpha2.crash.txt](https://github.com/prusa3d/PrusaSlicer/files/4000916/PrusaSlicer-2.2.0-alpha2.crash.txt)

#### Project File (.3MF) where problem occurs

[gifttag_crash.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4000917/gifttag_crash.3mf.zip)

lukasmatena, 2019-12-26:
This is clearly related to #3371. Apparently it is not Catalina specific...

lukasmatena, 2020-01-02:
I'll close this issue as a duplicate of #3371. The cause is surely the same and the issue tracker is cluttered too much. Hope that's ok.

closed 2020-01-02 by lukasmatena