prusa3d/PrusaSlicer #3395: [ISSUE] false generation of brim when supports are used

Moorviper, 2019-12-26:
[image]
[Inhaler v3.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4002763/Inhaler.v3.3mf.zip)

### Version
2.2.0 alpha 2 and earlier

### Operating system type + version
High Sierra 10.13.6

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Wrong placement of the Brim

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

bubnikv, 2021-02-23:
Fixed with 055d232

closed 2021-02-23 by bubnikv

bubnikv, 2021-02-23:
Current master prints supports over brim.

[image]

Implemented with 055d232