prusa3d/PrusaSlicer #3396: Phantom Support columns interfere with brim

tuxmobil, 2019-12-26:
### Version
2.2.0-alpha2+win64
_Version of PrusaSlicer used goes here_

### 3D printer brand / version + firmware version (if known)
MK2.5S

### Behavior
Brim around object has a gap, maybe because the slicer wanted to build support columns there, which I didn't want.
The aim was to have supports ONLY inside the object, without extending outside of the object. This already was not straight forward as the slicer always wanted to extend them outside of the object. (This eventually would be a problem because the final object would need the whole print bed space).

#### Project File (.3MF) where problem occurs
[image]

[Halter für Schrank.zip](https://github.com/prusa3d/PrusaSlicer/files/4003046/Halter.fur.Schrank.zip)

bubnikv, 2021-02-23:
Fixed with 055d232

closed 2021-02-23 by bubnikv

bubnikv, 2021-02-23:
Current master prints the supports over brim.

[image]

Implemented with 
055d232