prusa3d/PrusaSlicer #3397: multiple support material (MSM) - Feature Request

ReactSub, 2019-12-26:
- Waiting on my machine
- Windows 7 Pro 6.1.7601 Service Pack 1 Build 
- Prusa 3MKS w/ MMU 2

### Behavior
Soluble interface is great for saving material. But I'd like to be able to use an even less expensive material as the bulk support. 

Consider a three material setup that keeps temperatures close together (~240 C):

MatterHacker PRO series nylon $121/kg
Ionic high temp soluble Hybrid $264/kg
3DXMAX® ASA $38/kg

With the current soluble interface option it's cheaper to use the nylon as the bulk of the support (that's the point of the feature), since it's half the cost of the Hybrid. But by allowing the print body, soluble interface, and bulk support to be **three** different materials, the savings are accelrated.

**FOR EXAMPLE**:

Let's assume a small 25g print with 25g in support. Of the support, 20g is bulk support and 5g is interface. We'll assume a wipe tower with the exact same weights--so 100g total for all filaments used in total.

The material value of the final print is $3.02 in nylon.

Total cost _for just the tower and support material_ 
- with existing support method is $10.51
- using the proposed method waste is $7.19

That's a 1/3 reduction in cost of waste material for a 25g print. The simple math is that for a 250g print, it's a $30 material savings. 

_Is this a new feature request?_
Yes.

bubnikv, 2020-01-02:
Adding another material for the supports will trigger another tool change thus another wipe into the wipe tower. I am not quite sure whether you would save any material at the end of the day.

ReactSub, 2020-01-11:
That's an important point. However, I am quite sure. 
I have a model in mind that my method would successfully apply to. 

In the same way that sometimes a tower is three times the weight of model, and sometimes it's 50%, and can be anything between, we can't conclusively blanket that this wouldn't be useful. It's a feature request, after all. 

You did get me rethinking, and now I realize it could be even better. Here's how:
The across the board fix to what you outline is using the support structure as the wipe tower as "wipe support to (soluble) interface" in the same way as wipe-to-infill does--effectively using the support block as a wipe tower. It doesn't matter if the support structure dissolves in odd places during post. It's already throw-away.

closed 2026-09-29 by prusaslicer-bot[bot]