prusa3d/PrusaSlicer #3399: Default extrusion width not taking into account layer height

dewi-ny-je, 2019-12-26:
The default extrusion width in PrusaSlicer is fixed: 1.125 times the nozzle diameter.
However, for the reasons explained in https://manual.slic3r.org/advanced/flow-math an extrusion width matching flow speed and head speed would be preferable.

The difference can be seen in https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/Flow.cpp and in https://github.com/slic3r/Slic3r/blob/8e5a9cba71e8408c8fbcfa91f7c274e224e7f7a2/xs/src/libslic3r/Flow.cpp

The extrusion width from Slic3r and Prusa Slicer matches at about 50% layer height. Any other choice will produce sub-optimal extrusion width.

What are the reasons for the simplified calculation?

bubnikv, 2020-01-02:
> What are the reasons for the simplified calculation?

To match the defaults to our hand tuned profiles. The original code from upstream slic3r only caused troubles.

bubnikv, 2020-02-04:
Explained, closing.

closed 2020-02-04 by bubnikv