prusa3d/PrusaSlicer #3400: PrusaSlicer crashes and closes

Jman9912, 2019-12-27:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Everytime I open prusaslicer, it closes not long after. I don't get any error message or anything, it just closes. This problem recently started. I didn't change anything on my computer. I don't remember if it started after I updated prusaslicer or before. I have an acer nitro 5 with a Geforce 1050 graphics controller. I have restarted the computer, uninstalled prusaslicer, tried compatibility mode, tweaked my graphics settings, and everything else I can think of. I don't know how to find the error log. This is extremely annoying, any help would be greatly appreciated! I can't even slice anything to print.

bubnikv, 2020-01-02:
Do you have the same issue with PrusaSlicer 2.2.0-alpha2?

TNgineering, 2020-01-21:
I'm having this same issue. I've tried moving the MESA folder .dll into the main PrusaSlicer folder and it's still crashing

Jman9912, 2020-01-22:
Try opening as administrator

bubnikv, 2020-03-15:
Dreaded nahinmic issue?
https://github.com/prusa3d/PrusaSlicer/issues/2522

rtyr, 2020-05-04:
Is this issue still valid?

rtyr, 2020-12-03:
No response. Closing.

closed 2020-12-03 by rtyr