prusa3d/PrusaSlicer #3401: Slicer missing part of model

gknops, 2019-12-27:
### Version
2.1.1+ 201912101508

### Operating system type + version
MacOS Catalina (10.15.4 (19E193))
### Behavior
Attached STL file slices incorrectly (fine in Cura), see attached screenshot.

[InkWell3.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4004117/InkWell3.stl.zip)
[image]

gknops, 2019-12-27:
Screenshot did not make it, trying again:
[image]

gknops, 2019-12-27:
3rd time is the charm?
[image]

JohnOCFII, 2019-12-27:
It's a problem with the model.  Ideally, PrusaSlicer would notice that there was a model problem, but it didn't.  I've gotten in the habit of running my models through the free NetFabb service if I get a weird slicing issue.  So far, fixes them all.

[InkWell3_fixed.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/4005607/InkWell3_fixed.stl.zip)

gknops, 2019-12-27:
Thanks for the quick response, appreciated! I'll try to contact OnShape about this. BTW Cura sliced this without any issues.

closed 2026-09-29 by prusaslicer-bot[bot]