prusa3d/PrusaSlicer #3402: 2.2.0-alpha2 does not compile due to wxString vs char* error

bnavigator, 2019-12-27:
Similar to #2668 the new alpha-version does not compile on OpenSUSE distributions because of char* to wxString mismatch. ``ImGui::TextUnformatted()`` does not like the provided `wxString`.

https://github.com/prusa3d/PrusaSlicer/blob/220e2dbfec58cbbcff548d8211a17f0dcbe32aab/src/slic3r/GUI/GLCanvas3D.cpp#L280

From [the buildlog](https://build.opensuse.org/package/live_build_log/home:bnavigator:branches:science/PrusaSlicer/openSUSE_Tumbleweed/x86_64):

```
[ 2104s] [ 65%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GLCanvas3D.cpp.o
[ 2104s] cd /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build/src/slic3r && /usr/bin/c++  -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_LOCALE_DYN_LINK -DBOOST_LOG_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBNEST2D_GEOMETRIES_clipper -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_UNICODE -D__WXGTK__ -DwxDEBUG_LEVEL=0 -DwxUSE_UNICODE -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build/src/platform -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/clipper -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/polypartition -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/hidapi/include -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build/src/libslic3r -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/libnest2d/include -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/include -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/miniz -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/glu-libtess/include -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/qhull/src -isystem /usr/include/eigen3 -isystem /usr/lib64/wx/include/gtk2-unicode-3.1 -isystem /usr/include/wx-3.1 -isystem /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/libigl  -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG -fext-numeric-literals -Wall -Wno-reorder -pthread -O2 -g -DNDEBUG -fPIC   -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -pthread -std=gnu++17 "-Winvalid-pch" "-include" "/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build/src/slic3r/libslic3r_gui_pch/pchheader.hpp" -o CMakeFiles/libslic3r_gui.dir/GUI/GLCanvas3D.cpp.o -c /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp
[ 2107s] In file included from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/libslic3r/PrintBase.hpp:14,
[ 2107s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/libslic3r/Print.hpp:4,
[ 2107s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/BackgroundSlicingProcess.hpp:12,
[ 2107s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp:19:
[ 2107s] /usr/include/tbb/mutex.h:21:140: note: #pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.
[ 2107s]    21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
[ 2107s]       |                                                                                                                                            ^
[ 2109s] In file included from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/Gizmos/GLGizmoBase.hpp:6,
[ 2109s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/Gizmos/GLGizmoMove.hpp:4,
[ 2109s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/Gizmos/GLGizmos.hpp:29,
[ 2109s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp:2:
[ 2109s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp: In member function 'void Slic3r::GUI::GLCanvas3D::LayersEditing::render_overlay(const Slic3r::GUI::GLCanvas3D&) const':
[ 2109s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/I18N.hpp:2:45: error: cannot convert 'wxString' to 'const char*'
[ 2109s]     2 | #define _(s)    Slic3r::GUI::I18N::translate((s))
[ 2109s]       |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~
[ 2109s]       |                                             |
[ 2109s]       |                                             wxString
[ 2109s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp:280:32: note: in expansion of macro '_'
[ 2109s]   280 |         ImGui::TextUnformatted(_(L("Higher print quality versus higher print speed.")));
[ 2109s]       |                                ^
[ 2109s] In file included from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/ImGuiWrapper.hpp:7,
[ 2109s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GUI_App.hpp:8,
[ 2109s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/OptionsGroup.hpp:14,
[ 2109s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/BedShapeDialog.hpp:6,
[ 2109s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/Tab.hpp:31,
[ 2109s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp:23:
[ 2109s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/imgui/imgui.h:322:57: note:   initializing argument 1 of 'void ImGui::TextUnformatted(const char*, const char*)'
[ 2109s]   322 |     IMGUI_API void          TextUnformatted(const char* text, const char* text_end = NULL);                // raw text without formatting. Roughly equivalent to Text("%s", text) but: A) doesn't require null terminated string if 'text_end' is specified, B) it's faster, no memory copy is done, no buffer size limits, recommended for long chunks of text.
[ 2109s]       |                                             ~~~~~~~~~~~~^~~~
[ 2110s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp: In member function 'int Slic3r::GUI::GLCanvas3D::_load_print_object_toolpaths(const Slic3r::PrintObject&, const std::vector<std::__cxx11::basic_string<char> >&, const std::vector<Slic3r::Model::CustomGCode>&)::Ctxt::get_color_idx_for_tool_change(std::vector<Slic3r::Model::CustomGCode>::const_iterator, int) const':
[ 2110s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp:5458:32: warning: comparison of integer expressions of different signedness: 'size_t' {aka 'long unsigned int'} and 'int' [-Wsign-compare]
[ 2110s]  5458 |             if (number_tools() == extruders_cnt + 1) // there is no one "M600"
[ 2110s]       |                 ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
[ 2110s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp: In member function 'void Slic3r::GUI::GLCanvas3D::_load_gcode_extrusion_paths(const Slic3r::GCodePreviewData&, const std::vector<float>&)':
[ 2110s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/src/slic3r/GUI/GLCanvas3D.cpp:5952:71: warning: comparison of integer expressions of different signedness: 'const uint32_t' {aka 'const unsigned int'} and 'int' [-Wsign-compare]
[ 2110s]  5952 |                 if (is_selected_separate_extruder && path.extruder_id != m_selected_extruder - 1)
[ 2110s]       |                                                      ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
[ 2125s] make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:227: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GLCanvas3D.cpp.o] Error 1
[ 2125s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build'
[ 2125s] make[1]: *** [CMakeFiles/Makefile2:1431: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
[ 2125s] make[1]: Leaving directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.2.0-alpha2/build'
[ 2125s] make: *** [Makefile:144: all] Error 2
```

Viehwadschdiggl, 2019-12-27:
I've had similar compilation errors for several weeks on OpenSUSE Leap 15.0 using wxWidgets-3.1.1 from in the official OpenSUSE repo. Then I just gave it a try and installed wxWidgets-3.1.3 (by compiling from source) and since then PrusaSlicer compiled without errors.
I just updated to the latest commit 220e2db and again it compiled flawlessly.

I have no idea if the real solution was the newer wxWidgets version or maybe some issues with the pre-built version from the OpenSUSE repo - I just thought I'll let you know.

bnavigator, 2020-01-06:
Is there some automatic type conversion that does not work on the specific OpenSUSE build? How does compilation not fail with inconsistent types on other systems?

closed 2020-02-21 by lukasmatena

lukasmatena, 2020-02-21:
@bnavigator I believe that the reason is this commit: https://github.com/wxWidgets/wxWidgets/commit/e125c3b6573972ccfc06d228a7d5abd5306f73be , which enables that implicit conversion by default. wxWidgets in OpenSUSE apparently have it disabled. I have slightly amended and merged your fix. Thanks.

We should disable these conversions across PrusaSlicer codebase too, because it is a nasty pitfall. I did it locally and uncovered several more potential problems like this. Those should be fixed by 91cabe5832467cf546c4f000ceda9cc2c287a89e and f60c11776103ac7506174c4fafd23a89feba0f69. The public dependency scripts have not yet been updated to have the conversions disabled.

Thanks for pointing the problem out.

lukasmatena, 2020-02-27:
The unsafe conversions were disabled in PrusaSlicer cmake by 7c55674. It will hopefully fix this kind of issues for good.

commit a877147a (2020-02-21): fix #3402

Amended by lukasmatena: changed mb_str() to ToUTF8()