prusa3d/PrusaSlicer #3404: Spelling Mistake in Dialogue Box

ravingDIODE, 2019-12-27:
### Version
Version 2.2.0-alpha1+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

MK3S

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

There is a spelling mistake in the dialogue box when successfully removing a flash drive.
See screenshot here: http://prntscr.com/qggund

_Is this a new feature request?_

No.

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

[image]

spuder, 2019-12-27:
I"m not sure this is a prusa slicer problem since the string with the misspelling does not appear in this git repo. 'succeseful'

kocikdav, 2020-01-02:
Thank you for reporting. The typo will be corrected in next version.

closed 2020-01-02 by kocikdav