prusa3d/PrusaSlicer #3405: Plater texture changes after slicing

David-DL, 2019-12-27:
### Version
_2.1.1+win64-201912101512_

### OS version
_Windows 10 Home 1903_

### Behavior
* _Describe the problem_
After a fresh start of PrusaSlicer, after it finishes slicing, the texture of the plater gets distorted and stays like that until I close and open it again. After that the texture looks as it should again, but the behavior repeats after slicing.

* _Steps needed to reproduce the problem_
Open PrusaSlicer, add any model to the plater, select "Slice now".

* _Expected Results_
The plater texture displays always correctly.

* _Actual Results_
After slicing the plater texture gets distorted, and stays like that in all views, until the program is restarted.

#### Before slicing

[image]

#### After slicing

**_Video demo:_** https://youtu.be/P787tre4w_w

_*Note how the texture changes as I zoom in and out_
[image]

bubnikv, 2020-01-02:
Do you get this issue with PrusaSlicer-2.2.0-alpha2? @enricoturri1966 reduced the maximum size of the GPU texture, so this issue may be improved on low memory GPUs.

David-DL, 2020-01-03:
Hello.
Just tried with PrusaSlicer 2.2.0-alpha2 and I'm still seeing the same behavior.
Note that I'm using integrated graphics, a Ryzen 3 2200G.

[image]
[image]

In case this helps, but this is my GPU memory usage according to windows task manager. It stays mostly the same before and after slicing:

[image]

bubnikv, 2020-01-03:
This is certainly weird. It seems the high resolution texture of the bed is
all right all the time, but the low resolution texture gets messed up. I
can always recommend to try to update your graphics drivers, or if you have
a dual graphics card setup, to try to switch to the other one. Without
access to your system, it is really hard to debug your issue, especially if
the issue is rare (it happens on a single computer in the world, or on just
a handful of them).

David-DL, 2020-01-10:
Hello, sorry for the late answer.
Thanks @bubnikv! I checked and the problem indeed was related to an outdated graphics driver.

I'm using a Ryzen 2200G (integrated graphics, no discrete GPU) and updated the integrated Radeon Vega 8 drivers from version 24.20 to 26.20. It worked perfectly after that.

Thanks!

closed 2020-01-10 by David-DL

bubnikv, 2020-01-10:
thanks for heads up.