prusa3d/PrusaSlicer #3406: Feature request - Multi-Part Printing

tedukes, 2019-12-27:
_Is this a new feature request?_
Yes
Looks like Prusa Slicer is close to having this except for a retraction setting.

Using the slider after after slicing you can pick a point and add a setting.

This is what I'm trying to explain: https://www.simplify3d.com/support/articles/multi-part-printing/

BeatSlayer, 2019-12-27:
The feature already exists, it's called "Sequential Printing" and sits under the "Output Options" of your print profile.

tedukes, 2019-12-27:
Thanks, but I don't think I explained it too well.

Basically, you have two or more different parts with different settings, that print simultaneously but are two separate processes, then combined and sliced. Looks like using sequential printing, it prints one part then the next.  How would you set different settings for different parts using sequential printing?

BeatSlayer, 2019-12-27:
Using modifier meshes, but I see the limitation of you want to use totally different print profiles.

tedukes, 2019-12-28:
Thanks, I have the MMU2S. I can set retractions for various extruders but there are not filament or print settings for each extruder. 

I am using an Ender 3 Pro with a SKR 1.3 board and Marlin 2.0 

I think this video explains what i'd like to see if possible:  https://www.youtube.com/watch?v=Wadvbq0ov4U

What got me on this was a thing from thingiverse. I tried to run the gcode supplied. Made some changes for the start gcode but it failed miserably: https://www.thingiverse.com/thing:3236564

rtyr, 2021-02-04:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3635. Closing.

closed 2021-02-04 by rtyr