prusa3d/PrusaSlicer #3407: Astrobox Printer Host Upload

spuder, 2019-12-27:
PrusaSlicer has support for Octoprint and Duet

[image]

Add support for AstroPrint

https://www.astroprint.com/

API Documentation location

https://www.astroprint.com/developers

https://developer.astroprint.com/docs

bubnikv, 2020-01-24:
Why oh why? We have just found out that it works with OctoPrint upload protocol, so there is no need for new code.

spuder, 2020-01-25:
Several reasons:

1. Makes is clearer that not just octoprint works with slicer, there are other tools out there
2. There are two apis. AstroBox & AstroPrint. Not both are compatible, this would pave the way to support both in the future
3. Changing the box to say "Octoprint/Astrobox" just didn't seem like the right solution, but if you prefer that approach, I could be convinced.

bubnikv, 2020-02-04:
ok, let's keep it integrated as it is.
Thanks for contributing.
Closing.

closed 2020-02-04 by bubnikv