prusa3d/PrusaSlicer #3408: Wipe Tower ends with floating lines

tbrummel, 2019-12-27:
2.2.0 Alpha 2
Not introduced in this version - exists in previous versions as well
Windows 10

I was playing with the new sparse wipe tower feature in this alpha (thank you! @lukasmatena, I believe) and noticed that the last two traces of a wipe tower are hanging in the air.  This has been the case for some time and is NOT related to sparse wipe towers. 

I'm guessing this is a bug? Why would we want to print in the air as the print ends?  I'm not advocating that we keep printing the wipe tower so there is somewhere to deposit this last extrusion :-)  I don't think it needs to happen.  There is no tool change - the print is done.

Standard slice (note the hanging extrusion at the top of the tower):
[image]

With the sparse wipe tower feature enabled (nice! - except for the floating extrusion):
[image]

jmd73, 2019-12-27:
This has been discussed numerous Times before. The FAQ says:

Why there is an extrusion hanging in the air over a multi material wipe tower?
That is the final ramming before the filament is pulled from the nozzle and parked into the filament switcher. See #1355 for further details.

tbrummel, 2019-12-27:
Thanks - I did search and didn't run find the issues discussing this.  I'll check 1355 for details.

closed 2019-12-27 by tbrummel