prusa3d/PrusaSlicer #3409: Acceleration issue

tonn0297, 2019-12-27:
### Version
2.2 alpha 2

### Operating system type + version
Windows 10

### Behavior
if you set Acceleaton control default to 0, it dont dissable the acceleration control. 
it still takes the numbers in the grayed out fields.
it should take machine limits if marlin else it should not do a m204.
[image]
[image]

boilerbots, 2021-07-13:
Days of debugging and I was burned by this as well. No matter what "flavor" output I select my gcode always has M204 commands with values from disabled fields. I just downloaded 2.3.2 and this bug is still causing problems.

boilerbots, 2021-07-13:
also if you change the Title to include M204 then perhaps more people will find it.

boilerbots, 2021-07-13:
To add some details because I went around in circles trying to figure this out because in the Klipper discord someone else had this problem and mentioned that they where able to disable the **M204** codes.

### What seemed to fix the problem:
Set the default acceleration to a non-zero number so the other entries are enabled. Then set all of them to a non-zero number. Then I started at the top and set each one to zero and ultimately set the default back to zero an now M204 does not show up in my gcode.

lukasmatena, 2021-10-20:
This should be fixed after I merged #6743, thanks @jschuh. Closing.

closed 2021-10-20 by lukasmatena