prusa3d/PrusaSlicer #3410: Blob at beginning of line, but none with Simplify3d

rackley096794, 2019-12-27:
### Version
2.2.0-alpha2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S 3.7.2

### Behavior
* Problem: At the _beginning_ of each layer line a blob is produced.
* Repro: Print 20mm x 20mm single wall thickness test cube

I have been troubleshooting this by changing various settings and none have had any impact on the blob size.  These include:  Print speeds, linear advance K-factor, retract distance, retract speed, z-hop distance, retract (or not) on layer change, wiping/not wiping, etc.  None have any affect on the blob whatsoever.

For troubleshooting purposes I printed the same cube in Simplify3d with the same settings as PrusaSlicer (all manually transferred) and low and behold, no blob at all.  

The PrusaSlicer cube on the left has an aligned seam.  The Simplify3d cube on the right does not have an aligned seam, but there are no blobs anywhere on the cube (ignore the small layer warp, that was my bad with the first layer, totally not related to this issue).  In fact the layer transition is so good I can't even tell where the seam is at all.

.3mf and gcode output from from both PrusaSlicer and Simplify3d are attached:
[line start blob.zip](https://github.com/prusa3d/PrusaSlicer/files/4005934/line.start.blob.zip)

[image]

BeatSlayer, 2019-12-27:
The layer transition on the S3D printed object it's most certainly happening on a corner, you can try to use the "Nearest" option for the seam position on PS.

rackley096794, 2019-12-27:
Yes looking at the S3D gcode and printed part I can locate the corner, although it is very difficult - it is almost totally seamless.

rtyr, 2020-06-18:
I don't see any issue here. Those cubes have different seam position. Set seam position to "nearest" to achieve similar seam position. It will be always significantly less visible in the corner.

btw. not so important, but print speeds are different in your gcodes (15mm/s in PS vs roughly 9mm/s in S3D gcode). Closing.

closed 2020-06-18 by rtyr