prusa3d/PrusaSlicer #3411: Top solid layer and Variable layer height feature

vsky279, 2019-12-27:
### Version
Version 2.2.0-alpha1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK2S

### Behavior
The following behavior can be potentially considered a bug. Anyway when someone is not in detail familiar with the Variable layer height feature the result can be surprising.

Layer height is set to 0.2 mm, Solid layers Top set to 4.

Let's use Variable layer height feature and let's assume that the functionality sets the top layer height to highest detail, e.g. 0.07 mm.

Normally with 0.2 mm detail, one assumes that the total top solid layers height is 4x0.2=0.8 mm while with the conditions described above it will be only 4x0.07=0.28 mm. I admit I was surprised by this behavior. 

I would assume the correct behavior would be that the number of top solid layers is adjusted to lead to height of 0.8 mm.

Area5142, 2019-12-28:
Version: 2.2.0-alpha1+Linux64, 2.2.0-alpha2+Linux64
OS: Linux Mint 19.3

I observed the same behaviour and ended up with print that had holes in the top layer due to very thin layers.
My initial assumption was that the top and bottom thickness was preserved, even with variable layer height.
"Layers and perimeters"->"Ensure vertical shell thickness" seems not to correct the problem.

bubnikv, 2020-01-02:
I would say this is a duplicate of 
https://github.com/prusa3d/PrusaSlicer/issues/156

bubnikv, 2020-02-05:
Implemented with 495a71e. It will be part of the upcoming PrusaSlicer 2.2.0-alpha4 release.

The two new config keys top_solid_min_thickness / bottom_solid_min_thickness define a minimum vertical shell thickness. The top shell thickness is calculated as a maximum of (sum over top_solid_layers * their respective layer heights) and top_solid_min_thickness, the bottom shell thickness is calculated similarly.

The results of the formula above are shown at the Print parameter page below the two new values to hint the user about the interaction of the old versus new config values.

top_solid_min_thickness has no meaning if top_solid_layers is zero,
bottom_solid_min_thickness has no meaning if bottom_solid_layers is zero.

Closing.

closed 2020-02-05 by bubnikv