prusa3d/PrusaSlicer #3413: Empty region of model filled during slicing

antonok-edm, 2019-12-28:
### Version
2.1.1

### Operating system type + version
Arch Linux

### 3D printer brand / version + firmware version (if known)
Not applicable

### Behavior
A valid STL file is rendered correctly in the "3D Editor View" of Prusa Slicer, but produces incorrect G-code when sliced.

The STL file describes a simple rectangular box with a hole in the top. I was able to generate the STL file by exporting the following model from OpenSCAD version 2019.05:

```scad
TOLERANCE = 0.05;

DIM_X = 100 - TOLERANCE;
DIM_Y = 50;

WALL_THICK = 1;
WALL_HEIGHT = 20;

translate([-DIM_X + TOLERANCE, 0, 0]) union() {
    cube([DIM_X, DIM_Y, WALL_THICK]);

    translate([0, 0, WALL_THICK]) linear_extrude(WALL_HEIGHT) union() {
        square([WALL_THICK, DIM_Y]);
        square([DIM_X, WALL_THICK]);
        translate([0, DIM_Y - WALL_THICK]) square([DIM_X, WALL_THICK]);
        translate([DIM_X - WALL_THICK, 0]) square([WALL_THICK, DIM_Y]);
    }
}
```

The issue is reproduced using default Print, Filament, and Printer settings (`0.20mm NORMAL`, `Generic PLA`, `Original Prusa i3 MK2S`).

This screenshot shows the expected appearance of the model in the "3D Editor View".
[image]

This screenshot shows the incorrect sliced version of the model in the "Preview".
[image]
Scanning through the layers of the sliced result reveals that the inside of the model is populated with `Internal Infill`, as if the original model were a solid rectangular block.

#### Project File (.3MF) where problem occurs
The following archive includes the relevant OpenSCAD, STL, and 3MF files: [slicer-bug.tar.gz](https://github.com/prusa3d/PrusaSlicer/files/4006722/slicer-bug.tar.gz)

BerndJM, 2019-12-28:
Apart from the fact that you construct such a simple box very complicated (with a simple difference of 2 cubes you can do the same) it looks like some rounding errors irritate the slicer.

Your TOLERANCE = 0.05 creates vertexes with 0.0499969 or 0.0499998 or similar in the generated STL.
I could be wrong, but I guess that OpenScad is screwing up somewhere by defining DIM_X = 100 - TOLERANCE and in the first translate again the + TOLERANCE.

As soon as you leave out the first translate, the box will be sliced correctly and these "crooked" vertexes disappear in the generated STL ...

antonok-edm, 2019-12-28:
> you construct such a simple box very complicated (with a simple difference of 2 cubes you can do the same)

Indeed, this is not a sane way to model a rectangular box. My original model was more of an L-shaped enclosure with a flange; I ended up with this model by removing any geometry that had no effect on the issue.

> it looks like some rounding errors irritate the slicer

Agree that it appears to be some sort of rounding/precision issue given that changing `TOLERANCE` slightly or removing the top-level `translate` resolves the empty space as expected. Using that knowledge I was able to workaround the issue in my own model, but figuring it out took quite a bit of effort!

> I could be wrong, but I guess that OpenScad is screwing up somewhere by defining DIM_X = 100 - TOLERANCE and in the first translate again the + TOLERANCE.

I'm hesitant to blame OpenSCAD, given that the model still renders hollow in Prusa Slicer (and a few other STL viewers) before it is sliced. I'd at least expect the two views to match up.

adamsiem, 2020-04-12:
I have experienced a similar issue with printing a tube object. An extra layer of infill on layer 7 of 160. This is in Prusa 2.2.0. I described it here and provided models and screenshots: https://forum.prusaprinters.org/forum/prusaslicer/turn-off-bridge-infill/#post-207278

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo