prusa3d/PrusaSlicer #3414: Bridge pattern not correct

bgiot, 2019-12-28:
### Version
2.1.1+-2019121011508

### Operating system type + version
Mac OS Catalina 10.15.2

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
The bridge pattern is not in the correct direction. 
Using this sample https://www.thingiverse.com/thing:3092168, 
you can see on the screenshot that it would be better to have the bridge first layer not in zigzag
It seems that bridge is done always in the same direction instead of taking into account the more accurate direction.
[image]

NovaViper, 2019-12-29:
I have the same issue, I was printing a phone case that had stripes in it, only 4 of the sides did an up and down motion while the 5th stripe did a straight line along the direction of the strip, which caused it to fail a little bit

NovaViper, 2026-09-18:
How do we build the alpha? I know for sure it's still present in v2.9.6