prusa3d/PrusaSlicer #3415: Display file name in title of the window

chatelao, 2019-12-28:
### Version
PrusaSlicer
Version:   2.1.0+win64
Build:     PrusaSlicer-2.1.0+win64-201909160915

### Operating system type + version
Windows 10

### Behavior
* _Steps needed to reproduce the problem_
  * Open a ".3mf" file or save a project to ".3m"
* _Expected Results_
  * The filename is displayed in the title bar
* _Actual Results_
  * No filename is displayed in the title bar

Is this a new feature request? - Yes

bubnikv, 2020-01-02:
Works for me:

[image]

chatelao, 2020-01-02:
Hi Vojtěch, I just recognize the failed pattern matching in my brain, due to the missing ".3mf" extension. Would be great to be added. I just read a very, very, very long title in search of the extension :-).

Happy new year to Praha from Bern

bubnikv, 2020-01-02:
I am not sure whether everyone will be happy if we show the .3mf extension in the application title. To me it is an excess information, it is the project name that is shown in the title.

chatelao, 2020-01-02:
To be hoest: I would say, the length of the Software Name ("PrusaSlicer-2.1.0+win64-201909160915 based on slic3er") excessive, as I don't need it 99% of the time.

"test.3mf - Prusa Slicer" would be great (inspired by Acrobat Reader)

My problem is basically solved, thank you very much.

bubnikv, 2020-01-02:
>  To be hoest: I would say, the length of the Software Name
("PrusaSlicer-2.1.0+win64-201909160915 based on slic3er") excessive, as I
don't need it 99% of the time.

That's true

bubnikv, 2020-02-19:
The latest alpha shows a shorter application description.
Closing.

closed 2020-02-19 by bubnikv

chatelao, 2020-02-19:
Thanks to Praha - I will check out the next public release.